luc-github/Repetier-Firmware-4-Davinci #250: Autobed level, Filament sensor not functioning. 2.0a version 1

razor-keen, 2017-05-29:
I flashed a duo last night, with little troubles I was able to get it functioning. A curiousity was, after the initial flash, there was no sound and nothing functioned other rhan the lcd. After selecting load failsafe, everything started to act right. 
When auto level is selected, the process starts, but when the extruder pin touches the contact point, it does not register,  causing a fail and ending the process. Is there a way to test the function of the circiut? maybe the circiut is open somewhere between the extruder and the board.
Also, I had to disable the filament sensor due to getting a no filament in extruder 1 message. It was functioning properly before the flash. 
I ran out of time and have not tested any other functions. Thanks in advance.

Additional question, is there a guide to setting up repetier host for use?

luc-github, 2017-05-29:
Yes to make it works, after flash, you must load fail safe as wrote in readme

Probe and plates need to be clean to work properly, the command to test the probe is : G31 
https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/src/ArduinoDUE/Repetier/Repetier.ino#L49

Filament sensor is tested as soon as the extruder is heated, so check your staring code to be sure you heat only the extruder you use
also the sensor detects no filament if the little board is removed, so be sure it is connected properly

The repetier host settings are available on several forums - I do not use it, so I cannot help

razor-keen, 2017-05-29:
Id actually like to just use cura. I cant seem to get it connected to the printer.  I have both extruders loaded. The board is still connected. The probe is whistle clean. ill try sending the code, as you suggest here shortly..

razor-keen, 2017-05-29:
maybe you could suggest what you use?

razor-keen, 2017-05-29:
Im not sure exactly what I am looking for. The read out was zprobe state:L

luc-github, 2017-05-30:
I use S3D and I print from SD card (wifi SD card)
probe has 2 state L = Low, or H = High, I forget which one is when has contact and when not

razor-keen, 2017-05-30:
I will look again after work. I suppose if there is no change in state i have a problem with the printer i need to llook at. Or ill need to trace it back to get the pin, and check to see if it matches the code.

razor-keen, 2017-05-31:
i found another issue. my prints were offset by 9 mm i changed this in the epprom settings, but this also causes extruder 1 to not swipe on the wiper. Ive noticed the x offset for each extruder, but they are measured in steps. My question is, if i used -9 in x axis, do i remove nine from max x pos, or adjust both extruders offsets?

razor-keen, 2017-05-31:
I think its something with the print settings. sorry

luc-github, 2017-05-31:
About the 9mm shift yes should be your settings

about calibration of second extruder, position of first extruder offset must be 0 for x and y as it is the reference, the second extruder x / y offset must be calibrated as no printer is identical
current value are my printer values and I got them using this dual extrusion cone : https://www.thingiverse.com/thing:21773

When changing extruder offset you do not need to change any other setting as reference for this offset is first extruder position

this is trial / error calibration but it is not too long - you get good value when the cone is regular and no shift when color change

I do not know another way to do such calibration but it is efficient IMHO

Good luck

razor-keen, 2017-05-31:
The offset between the extruders is correct. My issue actually lies in the placement of parts on the bed. All parts end up 9 to the right, despite being centered in the slicer program. Ive now tried changing the offset in the printer settings, but it adds more problems. So far, I ve changed the x min to -9 and the z probe x offset to 9. The only thing left affected is the first extruder ends up too far left of the wiper. Can you tell me where the g-code is located within the Firmeware? I suppose it could be done another way, but if it will work right, that should be enough. Thanks a million. :)

luc-github, 2017-05-31:
to center your print in bed the settings are in slicer not in FW
or it means your xmin is not correct in EEPROM

in host do home X, then G1 X0, 
-> if it is in side of bed (at x=0) it means your slicer setting is wrong somewhere
-> if it is not in side of bed (at -9mm or 9mm) then adjust Xhome/Xmin accordingly in EEPROM

razor-keen, 2017-05-31:
I've reset eprom x min pos to 0 and sent g1 x0. No movement, because printer thinks its already at 0 (home). Changing this to -0 fixs the extruder placement for printing. However, if i run clean, or auto bed level, the machine will miss the wipers or probe pads by 9mm. I can correct the z probe adjustment by changing z probe offset x to 9 to compensate. I had done this last night, but today adjusted the x length, and everything seems better. I think there was another issue that arose from adjusting the length, but ill have to cross that bridge when i get there.

razor-keen, 2017-05-31:
* -9. Correction from earlier. -0 would have no effect

razor-keen, 2017-05-31:
Yes, looks like when we adjust the x min pos, it causes the unload, load function to park 9mm to the right of where it should be.
Also unloading, loading extruder 1 causes the heads to drag over the contact pads. There doesnt seem to be a G1 z1 to protect the heads. 
The only two I can clear through epprom is fix cleaning function from wrecking into the side by adjusting x max length, and auto bed level due to the z probe x offset. 
So it seems that changing the x min pos comes with a lot of heartache. 
It affects all other functions that use the home for positioning. If you could tell me where to find the auto load function I could make an adjustment that checks offset and corrects itself. Or it would be great to add it as an epprom setting.

luc-github, 2017-05-31:
every settings are in EEPROM and be careful to do a home after changing xmin, because if not you will mess up everything
loading/unloding go to xmin so if there is a shift - it just means you did not home before

so you mean at home position () doing home X),from X point of view, extruder is physicaly 9 mm far from bed ?

razor-keen, 2017-05-31:
I home before every action. 

When I mentioned the extruder not moving and being 9mm off of where it needed to be, it was with eeprom setting of zero. At that point, all printer actions worked great. 

I then changed X min pos to -9 and it fixed the printing issue, and the position of extruder when G1 X0 was sent.
However:

* It caused the x axis carrage to crash into the y axis during cleaning on the left side. 
*It caused autobed leveling to be off by 9mm. 
*It caused The extruder to stop short of being past the wiper on extruder 1.( Head stopped on right side instead of passing to left side into drip pan.) 
*Changing the x length by 9mm solved the extruder crashing, and changing the z probe x offset fixed auto bed leveling. 
*Using the load unload function after homing causes the heads to drag across metal pads on the back of the bed.

razor-keen, 2017-05-31:
The issue is definately in the firmware. 
I am incorrect in drscribing what is causing the x axis to crash. its the combination of the x min pos and z probe offset. they add way too much causing a crash. 
Load and unload works fine like this though. limiting the x length to correct the crashing moves the load position. 

Im thinking the cleaning is based off of the home instead of x min, but im not sure why adding z probe offset it making it worse. If there was a way to base it off x min pos it may fix it. 
As far as loading and unloading, the z axis needs to auto lower a tiny bit.

luc-github, 2017-06-01:
>The issue is definately in the firmware. 

well for the moment you lost me - between the issue title / the issue description and now you say you incorrectly describing issue but I do not know what issue exactly as some may come actually from your changes also.

can you describe clearly (and separatly if several at once) the issue(s)  you meet with default FW/settings ?

razor-keen, 2017-06-01:
Yes sir. i appologize of any miscomunications. I will try to write them all down, one by one and how each change affected them tonight. I will run each of the features to see how the changes affect them. If not today then before saturday. Thank you.

razor-keen, 2017-06-01:
Just curious. What about loading and unloading the left extruder? Is there a way to drop the table so it doesnt swipe across the pads? I know i can do it manually, then run the feature, but can it happen automatically? I never found where the machine instructions where in the arduino code. I thought i would recognize them if they looked like gcode.

luc-github, 2017-06-01:
yes it is possible - it was no need until now as no one raised the problem on 2.0A and this issue does not happen on 2.0
as there is already a work around but may be distance is not enough on 2.0A
https://github.com/luc-github/Repetier-Firmware-0.92/blame/master/src/ArduinoDUE/Repetier/Configuration.h#L1048-L1053

razor-keen, 2017-06-02:
Loaded failsafe in epprom
Home all axis
Send G1 Z3 - Bed lowers a bit
Send G1 X0 - No movement
*Send G1 X9 - Movement E1 slightly over bed
Send M100 - Correct positions, good
Select load left extruder - No Z movement, E1 moves to correct location
Select auto bed level - correct positions, good

Change eeprom setting -  [(-9) x min pos]
Home all axis
Send G1 Z3
Send G1 X0 - E1 moves over bed slightly, good 
Send M100 - cleaning moves to far to left on cleaning e1
Select load left extruder -  No Z movement, E1 moves to correct pos
Select auto level - Sends E1 to far to the right

Change Eeprom setting - [(9) Z probe offset x]
Home all axis 
Send G1 Z3
Send G1 X0 - Movement correct (slightly over bed)
Send M100 - Extruder crashes on left belt slips then swipe is good, moves back wrecks into home (me thinks it lost place when the belt slipped, otherwise it would be way to far to right) Total 18mm movement increase!
Home all axis
Send G1 Z3 - If I don't send this the extruders swipe across the contacts in the back...
Select load left extruder - E1 moves to correct position
Select Auto level - Movement correct

Change Eeprom setting - [(190) X max length ; I subtracted 9 from 199]
Home all axis
Send G1 X0 - Movement correct (slightly over bed) 
Send G1 Z3
Send M100 - Cleaning is correct again
Select load left extruder - Stops short of the drip bin (about 9mm)
Select Auto level - Movement correct. 

My suggestion is to make the cleaning function based off of the X max length but the x max length needs to recognize the negative value set in X min pos. Im not sure why adding [Z probe x offset] would add 9mm on top of the extra 9mm it gets from [x min pos]. This is a total of 18 mm movement increase, when it should be limited by X total length.   

This is most likely only on the duo due, to the cleaning function. I can fix the other two issues using eeprom settings. 

 Im still curious as to why there is no z down function on load extruder 1 ( left extruder ), ( also a duo specific issue ).

One option is to make the cleaning home and end configurable in eeprom. 

Or

If you could tell me specifically where the hard code instructions are in the arduino file ill try to help. If I knew the lines responsible for what values are referenced, maybe we can fix it there. Unless I didn't read correctly the link you posted was for the Menu options. 

I have a feeling that x max length is not taking into account negative numbers, and starts movements at zero up to max, ignoring lengths added by negative offsets. 

Thanks for being prompt and not sweeping me under the rug. Im not sure if its also a problem on the 2.0a version 2, (blue) but on the 2.0a version 1 (grey), Its a bit troublesome. I basically have to decide which function I like best and set the eeprom so it will work and take a lose on another. 

Again, thanks for your time.

Jason

razor-keen, 2017-06-02:
On leveling the bed, this feature works and sets a good matrix, but the left side seems to not get a ground on the pad. I corrected it by adding a jumper between the two. Is the voltage adjustable? It might help so that it can not be so sensitive to debris. It would also be nice to be able to increment it down or up at one point to add an offset for fine tuning. Sorry most have given up on the feature but I rather like it.

razor-keen, 2017-06-02:
Changes made in configuration.h
I've set the [FILAMENTCHANGE_X_POS] to -9
I've set the [FILAMENTCHANGE_Z_ADD] to 3 
I've set the [FILAMENTCHANGE_REHOME] TO 3
Changing these did not help the load function like I assumed they might. 
I read that the adds the x offset to get total length so I should be right by changing x max length in eeprom. 
As far as I can tell, there is not any actual movement instructions located in configuration.h. That is what I am looking for right now. 
I think I've found the cleaning instruction code in printer.cpp. Its getting to late. Tomorrow is a long day for me and I wont have time to work on this. The arduino code is new to me, but it looks pretty basic. If I can help I will. We will see what the weekend holds for me.

luc-github, 2017-06-02:
As I still do not know what are the problem(s) - you describe what you try but not what is the problem

I just guess you have issue with zprobe plate so I have added a +10 on Z without homing if Zpos is under 10 : https://github.com/luc-github/Repetier-Firmware-0.92/commit/f15ec540fd2f51bc717c2e8ff114af995e53bfdb

I hope it helps as it does not hurt

razor-keen, 2017-06-02:
there is a description of what problems arise after each change i made. I wrote change eeprom setting. then wrote about various changes that happened with the behavior of the machine. problems included extruder stopping short of drip pan, crashing into sides, and cleaning function not working. Notice how each change affected each function. There is not a setting that can make all three functions operate at the same time. Im sorry but that is as clear as i can be.

luc-github, 2017-06-02:
that is my point you do several  changes then describe problems 
how I know the problem is with default settings or due to your new changes ?
I am not by your side seeing what you are seeing / doing, what it is obvious for you, it is not for people reading your description  
and how to follow in so many points you raise ?

I do not want to be rude but trying to explain to you I cannot follow what you are doing

to report an issue the process is: 
1 - report the issue describing the issue with as much details as possible to make others to understand the problem
2 - give the reproduction steps to let people be able to reproduce the issue you have. simple and clearly
3-  you can give solution if you understand the code of course
4 - do the above for each issue separatly

I am sorry but I read again your comments and I am not following you :
>Loaded failsafe in epprom
Home all axis
>Send G1 Z3 - Bed lowers a bit
>Send G1 X0 - No movement
>*Send G1 X9 - Movement E1 slightly over bed
>Send M100 - Correct positions, good
>Select load left extruder - No Z movement, E1 moves to correct location
>Select auto bed level - correct positions, good

where is the issue in this description ?

so 
>Change eeprom setting - [(-9) x min pos]
>Home all axis
>Send G1 Z3
>Send G1 X0 - E1 moves over bed slightly, good
>Send M100 - cleaning moves to far to left on cleaning e1
>Select load left extruder - No Z movement, E1 moves to correct pos
>Select auto level - Sends E1 to far to the right

why changing xmin to -9 if previously no issue ?

and because I do not understand your issue, I cannot comment the change you are doing (-9) 
this may be not the right solution and moreover will bring problems if you do not changes others setting accordingly.
As said before, changing Xmin means your x length need to be adjusted as reference changed, current values are for X=0, and others settings like the plates position, and others are also impacted.
I gave an explaination before for AiO: https://github.com/luc-github/Repetier-Firmware-0.92/issues/80#issuecomment-131387275

please try the latest git  version, the modification you requested is in it, also apply fail safe settings to get the base to report the issues you may encounter

I do not say there is no issue but I need help from you to help you

razor-keen, 2017-06-02:
I changed x min to correct part positioning, as most slicers assume 0 is the beginning of the bed. 
This may be a mistake on my part?

razor-keen, 2017-06-02:
i will try to reword my post to the best of my ability soon. You are not being rude, no worries. In my post i added a description of what the problem was with the settings under it. The first being from failsafe settings. I edited the comment a bit to help clarify.

razor-keen, 2017-06-02:
I took lunch at the house today, started the printer with failsafe settings and it printed centered. I don't know what changed I have not changed anything except eeprom settings lately. maybe closing and reopening repetier host? I will try and replicate what went wrong. I have not reladed repetier with the changes you made I will try tonight.  

Only remaining issues is the extruder 1 sensor doesn't seem to function under repetier. 

Im not sure where to start with that one.

Also, the powersave feature, should it turn the light off after a time?

razor-keen, 2017-06-03:
I just got home worked 14 hours today, loaded the changes you made. eeprom settings still just fail safe, the bed moved down and no longer drags on back plates, print was centered, load/unload function stopped shortof expected position. The second try was correct as was third and fourth. then fifth try stopped short of expected position. Im going to bed, will sort it out later, I just wanted to thank you for adding the drop on load function.
Thank you, I really mean it. I was worried I would ruin the stock extruder tips forgetting to lower the bed first.

luc-github, 2017-06-19:
Any update ? or issue can be closed ?

closed 2017-06-27 by luc-github