luc-github/Repetier-Firmware-4-Davinci #251: Da Vinci 2.0 with 2 fans, Motors not Working, bed, extruders not heating

chrisdarragh, 2017-05-31:
Recently flashed my 2.0 with repetier, and nothing seems to be working. I've tried multiple times to reflash the firmware with no luck.

On the main screen all the temperatures show up as 0 degrees, and whenever I try to print, no motors start up. I have tried both printing a sample file from the printer and sending stl files to it with no prevail.

When trying to print one of the sample files, the percentage will load and say that its printing and nothing happens at all. On repetier host, however, I send the file and it gives me error M999.

Not sure if this is my fault of being very new to the 3D printing game, but I have tried non stop to the extent of my knowledge. Please help me out

chrisdarragh, 2017-05-31:
[Repetier.zip](https://github.com/luc-github/Repetier-Firmware-0.92/files/1041844/Repetier.zip)

luc-github, 2017-05-31:
well you have a 2.0a with jumper 37 right ?
 if no that would explain the 0 temperature -  or did you do a hardware modification ? like changing extruder ?

for motors not moving it looks like you forget the last step of readme : load fail-safe settings 

trying to print sample file with repetier is not working - they mean to be used with stock FW

chrisdarragh, 2017-05-31:
I have the 2.0a yes, and no hardware modification. I'm assuming loading fail safe is on the printer itself?

chrisdarragh, 2017-05-31:
Before compiling it gives me this message as well-

In file included from sketch\Configuration.h:185:0,

                 from sketch\Repetier.h:200,

                 from sketch\Commands.cpp:22:

sketch\pins.h:1425:140: warning: narrowing conversion of '128' from 'int' to 'const int8_t {aka const signed char}' inside { } [-Wnarrowing]

                         HEATER_0_PIN, HEATER_1_PIN, HEATER_2_PIN,FAN_PIN, E0_PINS E1_PINS E2_PINS TEMP_0_PIN, TEMP_1_PIN, TEMP_2_PIN, SDSS }

                                                                                                                                            ^

sketch\Commands.cpp:24:41: note: in expansion of macro 'SENSITIVE_PINS'

 const int8_t sensitive_pins[] PROGMEM = SENSITIVE_PINS; // Sensitive pin list for M42

                                         ^

closed 2017-05-31 by chrisdarragh