luc-github/Repetier-Firmware-4-Davinci #252: Issues flashing Davinci 1.0a to Repetier

Amdivns, 2017-06-11:
Hi all, 

Firstly - i'm admittedly a n00b when it comes to coding - however i'm quite able to follow instructions, and am a quick study.

For reasons unbeknownst to me - I can't seem to flash my printer over to Repetier and wanted to check I was following all the steps correctly.

luc-github, 2017-06-11:
yes variants.cpp and USBCore.cpp are not in arduin install directory

gpriaulx, 2017-06-27:
I am having the same problem with the 1.0A bought in 2016 (I think) using the latest release:
Bossac -p com17 -R -e -w -v -b Davinci-1.0A-v0.92.10rc2d.bin

Any advice?

gpriaulx, 2017-06-27:
I just had success!

I'm not sure if any of the following are critical steps, but I have been logging my actions for my own sanity and after:

1. Short J37
2. Turn on Printer (connected to PC)
3. Turn off after 10 seconds
4. Remove Short
5. Turn on 
6. Change COM configuration in Windows from 17 to 2.
7. Attempt to execute: Bossac -p com2 -R -e -w -v -b Davinci-1.0A-v0.92.10rc2d.bin
8. Observed failure: No device found on com2
9. Unplug printer from USB port
10. Re-plug printer
11. Verify COM2 setting
12. Attempt to execute: Bossac -p com2 -R -e -w -v -b Davinci-1.0A-v0.92.10rc2d.bin
13. Observe success
14. Load failsafe config and store on EEPROM

zipippino, 2017-06-27:
gpriaulx please explain it to me better.
Where do you have been putting the "bossac ...."  commands?
I have a Da vinci 1.0a (j37)
I Followed your steps , other 2 youtube guides steps , installed all arduino version suggested , i tried the master and the dev branch but still nothing. In arduino console output it says 

> Verify successful
> done in 36.848 seconds
> Set boot flash true
> CPU reset.

But the 3d printer is still stuck at 2 bars on the screen. What should i do?
Please help meee

Jay1011, 2017-07-13:
Hey, I have been trying to flash my printer for months now and all I want is for it to work. Id there anyway that I could find out what I'm doing wrong. I am 97% sure I'm not copying my files in the right location. Is there anyway I could send a video or screenshots of the process I follow and someone could walk me through it?

luc-github, 2017-07-13:
@Jay1011 the way to find the path to copy is in FAQ here : https://github.com/luc-github/Repetier-Firmware-0.92/issues/114
if you are not able to compile properly you can use the binary from here : https://github.com/luc-github/Repetier-Firmware-0.92/releases/tag/v0.92.10rc2d
and directly flash using bossac using proper port and target binary :
`bossac -p <your com port> -R -e -w -v -b <the binary for your printer>`

Jay1011, 2017-07-13:
Luc, I can't find bossac or any specified files. I just don't know none of them are in the specified paths,

luc-github, 2017-07-13:
bossac is there
`C:\Users\<your user id>\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.6.1-arduino`

Jay1011, 2017-07-13:
I've looked there and when I open "Appdata" all that is there on my laptop is a file labeled Microsoft

luc-github, 2017-07-13:
did you installed due module ? FAQ https://github.com/luc-github/Repetier-Firmware-0.92/issues/166

Jay1011, 2017-07-14:
Luc, It's Working!!! thanks so much for everything. Now I just need to learn how to use it XD

luc-github, 2017-07-14:
Ok I close issue as seems no feedback from original owner

closed 2017-07-14 by luc-github

eyerulul, 2017-12-03:
@gpriaulx, I had a question on one of your commit? Can you spare 10 minutes to explain? feel free to reach me at eyerulul@gmail.com 

Thank You!

CollinsCreations1, 2019-02-12:
Hello I've been having ALOT of trouble bouncing between forums and videos figuring out the hard way as to which method is outdated. I am trying to flash my XYZ Da Vinvi 1.0A printer with repetier host software so that I may print with 3rd party filaments and use the Cura slicer. I've tried every Arduino version including 1.8.0 and have always added the correct Board through manager and selected the native USB option. I have selected COM 4 as this is the printer connection and have entered 1 for both of the configuration options. I started by flashing the printer by shorting the J37 pins and achieved the 2 black bars. I have transferred the correct hardware files from the repetier folder into the arduino folder. When I verify the arduino code it says successful however every time I upload to the printer even if it results in the code saying "upload complete/successful" the USB becomes unrecognized by my computer and the only way to re-establish the printer as "COM 4" again is by reflashing by shorting J37 again. It's been an endless loop of failure for me and I can't quite find the solution. As I've read from above there is a "Bossa Procedure" however I can't quite seem to find anything to walk me through it. If anyone has any recccomendations, I'd very much appreciate it!

zipippino, 2019-02-12:
@CollinsCreations1
I did it with bossac method, with arduino method i couldn’t do it. You can follow @luc-github ‘s instruction above:

https://github.com/luc-github/Repetier-Firmware-4-Davinci/issues/252#issuecomment-462582986

CollinsCreations1, 2019-02-12:
Thanks @zipippino but Luc was actually able to help me with it on a seperate thread, appreciate the willingness!

zipippino, 2019-02-12:
Ok no problem. Just woke up and saw this thread reply in inbox :)