luc-github/Repetier-Firmware-4-Davinci #253: Help my 1.0 does not move

hugobmf, 2017-06-19:
Hi, I have just made the flash for my Davici 1.0 new borad (J37), before compiling the firmware I changed it to "1" in the configuration.
The display, buttons, lights and all the menus work fine, the temperature sensors are also ok.
The repeater software is communicating via USB but no X, Y, Z axis commands and when I send an STL via repeater it accepts but does not change the temperature of the bed and nozzle and no axis moves.

What can I do to fix this problem?

hugobmf, 2017-06-19:
More information, I installed Repeater 2.0.1 and clicking to warm the table and the spout it initiates the process and gives an error seconds later:
Msg - fatal: Heater / sensor failure printer stopped and heaters disabled due to this error. Fix error and restart qith m999.

luc-github, 2017-06-20:
about not moving issue : does home functions work from printer itself ?
about heating issue : does heating works  from printer ?
you can disable decouple test in configuration.h  to debug this, it is usually due to a bad/loose connector : `#define FEATURE_DECOUPLE_TEST 0`

hugobmf, 2017-06-20:
Thank you Luc for your help,
Oops I did not load fail-safe after flashing how do I do this?

Nothing works from the printer itself I tried to move the axes, print via sdcard, clean the nozzle to load the filament but nothing worked.
I changed FEATURE_DECOUPLE_TEST 1 to flashing again but nothing has changed.

Only the fun will turn on when I command the aquacer nozzle, but I have no heating in the nozzle and in the bed

luc-github, 2017-06-20:
>Oops I did not load fail-safe after flashing how do I do this?

go to settings menu of printer

`FEATURE_DECOUPLE_TEST 0` won't solve any problem but avoid the heater test to be raised if heating is erratic, but in your case I guess the issue is the missing failed-safe settings or send M502 then M500 from host

luc-github, 2017-06-20:
hugobmf wrote
>And amazing how things work when you read the manual, I went through the installation step by step and found it to be fail safe and everything seems to be working fine. kkk
My XZY is free!!!

Thank you Luc!!!

luc-github, 2017-06-20:
Good happy printing 😸

closed 2017-06-20 by luc-github