luc-github/Repetier-Firmware-4-Davinci #254: DaVinci 1.0A doesn't print

zipippino, 2017-06-27:
Hi, 
after a lot of issues i finally installed the custom firmware on my printer.
Now when i try to print something it does the correct procedure and moves of the extruder , but it doesn't warms up the bed and the extr. automatically before printing and if i do it manually and then print , it doesn't push the ABS out anyway.
Any help?

luc-github, 2017-06-27:
there is no automatic heating, everything come from your Gcode  - check your slicer settings, expecially the start code and /or the temperature you set

zipippino, 2017-06-27:
@luc-github Is there a guide or at least a Gcode-values table?

luc-github, 2017-06-28:
GCODE are standard repetier ones plus some specific I have introduced for Davinci
 the full list : https://github.com/luc-github/Repetier-Firmware-0.92/issues/105

for the guide - it is better to check your slicer documentation

zipippino, 2017-06-28:
@luc-github Ok, if i want to say at example, when i press the start print button on replier host,
set the extruder position on 0,0 , then set extruder temp on 230*C , then the bed temp at 110, then print.
Where should i put that codes?
I'm sorry if i'm wasting your time but i would appreciate a lot if you would give a little help at this noob that i am.
Thanks a lot

luc-github, 2017-06-28:
I do not use repetier host for a while as I use S3D , but in all case heating manually is not necessary, this is done by your slicer - which in repetier host can be cura or slic3r 
so as I said read the documentation is the best advice I can give - https://www.repetier.com/#documantation
learning curve can be long or short depending of how you are willing to read to learn

zipippino, 2017-06-30:
Thanks a lot i solved.

closed 2017-06-30 by zipippino