luc-github/Repetier-Firmware-4-Davinci #257: The G1 Z command does not work ...

businka, 2017-06-30:
XYZ Duo 2.0A 0.93.10rc2b
G1 Z0.800 F7800.000  does not work
G1 X1 Y1 Z0.800 F7800.000 work

The nozzle at the beginning of printing always cuts into the table, you have to edit the G code with your hands.

sample
; generated by Slic3r 1.2.9 on 2017-06-30 at 12:32:36

; external perimeters extrusion width = 0.40mm
; perimeters extrusion width = 0.67mm
; infill extrusion width = 0.67mm
; solid infill extrusion width = 0.67mm
; top infill extrusion width = 0.67mm

M107
M190 S90 ; set bed temperature
M104 S210 ; set temperature
G28 ; home all axes
G1 Z5 F5000 ; lift nozzle

M109 S230 ; wait for temperature to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
G1 **X1 Y1** Z0.800 F7800.000

Sporque, 2017-06-30:
Have you looked at the slic3r settings for Z-offset? Change that to a positive number in millimeters, and save it, and you won’t have to manually change it in the Gcode every time.

From: businka
Sent: Friday, June 30, 2017 7:46 AM
To: luc-github/Repetier-Firmware-0.92
Cc: Subscribed
Subject: [luc-github/Repetier-Firmware-0.92] The G1 Z command does not work... (#257)

XYZ Duo 2.0A 0.93.10rc2b
G1 Z0.800 F7800.000 does not work
G1 X1 Y1 Z0.800 F7800.000 work
The nozzle at the beginning of printing always cuts into the table, you have to edit the G code with your hands.
sample
; generated by Slic3r 1.2.9 on 2017-06-30 at 12:32:36
; external perimeters extrusion width = 0.40mm
; perimeters extrusion width = 0.67mm
; infill extrusion width = 0.67mm
; solid infill extrusion width = 0.67mm
; top infill extrusion width = 0.67mm
M107
M190 S90 ; set bed temperature
M104 S210 ; set temperature
G28 ; home all axes
G1 Z5 F5000 ; lift nozzle
M109 S230 ; wait for temperature to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
G1 X1 Y1 Z0.800 F7800.000

luc-github, 2017-06-30:
F7800 is too high for Z, I am not surprised it may not move, it is too fast so it loose so many steps that it does  not move as expected

_Sent from my LGE LG-H990 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

businka, 2017-06-30:
> Have you looked at the slic3r settings for Z-offset? Change that to a positive number in millimeters, and save it, and you won’t have to manually change it in the Gcode every time.

Z-offset I have installed - I add X1 Y1 to drop the bed

> F7800 is too high for Z, I am not surprised it may not move, it is too fast so it loose so many steps that it does not move as expected

what can be done? Speed of the Z axis?

luc-github, 2017-07-01:
Yes here the value I use on S3D
[image]

so you should get something like : `G1 Z5.00 F600`

businka, 2017-07-03:
I determined the problem.
It's not about speed. Z axis travel works with both F600 and F7800

The thing is that the Z-axis movement does not work if X = 0 and Y = 0.

luc-github, 2017-07-03:
I just send these commands and they work as expected:
`
G28 Z0
G1 Z5 F5000
G1 Z0.800 F7800.000
`

luc-github, 2017-07-03:
same for :
```
G28
G1 Z5 F5000
G1 Z0.800 F7800.000
```

luc-github, 2017-07-03:
So if you can share a small script that reproduce issue would great

businka, 2017-07-04:
Floating error.
I stopped printing yesterday and did not turn off the printer. And team
G28
G1 Z10
Works as it should.
I turned it off, turned on the printer and everything repeats
G28
G1 Z10
does not work
G28
G1 X1 Y1 Z10
working

This is repeated on reperier host 2.0.1 - direct connection and through repetier server

businka, 2017-07-04:
Stably repeated. Can i make a video

luc-github, 2017-07-04:
Sure, as I do same and do not have any issue - what is the baud rate you use on printer and host ?

businka, 2017-07-10:
I did not touch the speed. Were by default.
Checked, turned out to be different.
Repetier-server 115200
Repetier-host - 250000
Da vinci - 230400

Did all 230400
The problem persists
After switching on G28; G1 Z10; does not work

luc-github, 2017-07-16:
well if baud rates are different it won't work well for sure ...

what version of host are you using ?

businka, 2017-07-17:
host 2.0,1
server 0.85

luc-github, 2017-07-17:
do you have the log of repetier when sending commands ?

businka, 2017-07-20:
13:09:49.018 : N4873 G1 Y50*108
13:09:49.018 : x3
13:09:49.018 : X:-0.01 Y:0.00 Z:100.000 E:100.0000
13:09:59.720 : N4878 G1 X10 Y50*30
13:10:34.274 : N4890 G1 Y10*101

businka, 2017-07-20:
When the extruder is in the home position, it does not respond to commands on the same axis

businka, 2017-07-21:
Home
1. G1 Y50 -dont work
response
13:09:49.018 : x3
13:09:49.018 : X:-0.01 Y:0.00 Z:100.000 E:100.0000
2. G1 X10 Y50 work
3. G1 Y10 work

luc-github, 2017-07-23:
I have used Repetier Host 2.0.1 and enabled all in manual control to get all feed back and do not reproduce your issue - can you past the log as I do ?
```
12:06:18.833 : wait
12:06:19.834 : wait
12:06:20.234 : N18 G28*42
12:06:20.234 : ok 18
12:06:20.261 : SelectExtruder:0
12:06:20.262 : FlowMultiply:100
12:06:22.250 : busy:processing
12:06:24.256 : busy:processing
12:06:26.250 : busy:processing
12:06:26.284 : X:0.00 Y:0.00 Z:0.000 E:0.0000
12:06:26.284 : wait
12:06:44.248 : N19 G1 Y50*108
12:06:44.248 : ok 19
12:06:45.249 : wait
```

businka, 2017-07-30:
I'll turn on all options in the console on Monday and run G28, G1 Y50. I understood correctly?

luc-github, 2017-07-30:
yes , thank you

_Sent from my Lge LG-H990 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

businka, 2017-07-31:
After turning on everything works as it should.
The problem occurs after printing is stopped.

Log after a stop
On unsuccessful commands comes the answer "x3"
```
11:17:59.043 : N8013 G1 X73.622 Y62.749 E2.46094*89
11:17:59.558 : ok 8007
11:17:59.558 : ok 8008
11:17:59.558 : N8014 G1 X76.382 Y62.749 E2.57649*82
11:17:59.683 : ok 8009
11:17:59.683 : N8015 G1 E0.57649 F2400*54
11:17:59.683 : N8016 G92 E0*120
11:18:00.229 : ok 8010
11:18:00.229 : N8017 M105*25
11:18:00.353 : ok 8011
11:18:00.353 : N8018 G1 X76.342 Y59.892 F7800*122
11:18:00.353 : N8019 G1 E2 F2400*47
11:18:00.931 : ok 8012
11:18:01.055 : ok 8013
11:18:01.055 : N8020 G1 X73.663 Y59.892 E2.11214 F3600*54
11:18:01.648 : ok 8014
11:18:01.648 : N8021 G1 X73.704 Y57.035 E2.23175*90
11:18:01.757 : ok 8015
11:18:01.757 : N8022 G1 X76.302 Y57.035 E2.34047*88
11:18:02.350 : ok 8016
11:18:02.350 : N8023 M117 ETE/ 2h 33m 19s*42
11:18:02.350 : ok 8017
11:18:02.366 : T:229.29 /230 B:89.18 /90 B@:255 @:230 T0:229.29 /230 @0:230 T1:49.66 / 0 @1:0
11:18:02.366 : ok 8018
11:18:02.366 : N8024 M104 S0 T0*31
11:18:02.459 : ok 8019
11:18:02.459 : N8025 M104 S0 T1*31
11:18:03.052 : ok 8020
11:18:03.052 : N8026 M140 S0*89
11:18:03.177 : ok 8021
11:18:03.177 : N8027 M84*34
11:18:03.177 : N8028 M105*21
11:18:03.754 : ok 8022
11:18:03.863 : ok 8023
11:18:03.895 : ok 8024
11:18:05.891 : busy:processing
11:18:06.110 : N8029 M105*20
11:18:07.795 : TargetExtr 0: 0
11:18:07.810 : Printed filament:200.35m Printing time:1 days 12 hours 30 min
11:18:07.810 : ok 8025
11:18:07.810 : TargetExtr 1: 0
11:18:07.810 : ok 8026
11:18:07.810 : TargetBed: 0
11:18:07.810 : ok 8027
11:18:07.841 : T:228.26 / 0 B:91.87 / 0 B@:0 @:230 T0:228.26 / 0 @0:230 T1:49.88 / 0 @1:0
11:18:07.857 : ok 8028
11:18:07.857 : T:228.33 / 0 B:91.83 / 0 B@:0 @:0 T0:228.33 / 0 @0:0 T1:49.88 / 0 @1:0
11:18:07.857 : ok 8029
11:18:07.857 : T:228.33 / 0 B:91.83 / 0 B@:0 @:0 T0:228.33 / 0 @0:0 T1:49.88 / 0 @1:0
11:18:08.840 : wait
11:18:09.152 : N8030 M105*28
11:18:09.152 : ok 8030
11:18:09.152 : T:228.74 / 0 B:91.49 / 0 B@:0 @:0 T0:228.74 / 0 @0:0 T1:49.88 / 0 @1:0
11:18:10.150 : wait
11:18:11.149 : wait
11:18:12.147 : wait
11:18:12.209 : N8031 M105*29
11:18:12.209 : ok 8031
11:18:12.209 : T:230.11 / 0 B:90.15 / 0 B@:0 @:0 T0:230.11 / 0 @0:0 T1:49.88 / 0 @1:0
11:18:13.208 : wait
11:18:14.206 : wait
11:18:15.205 : wait
11:18:15.267 : N8032 M105*30
11:18:15.267 : ok 8032
[... 275 lines cut ...]
11:19:28.649 : N8058 M105*18
11:19:28.649 : ok 8058
11:19:28.649 : T:156.87 / 0 B:82.00 / 0 B@:0 @:0 T0:156.87 / 0 @0:0 T1:48.56 / 0 @1:0
11:19:29.648 : wait
11:19:30.646 : wait
11:19:31.645 : wait
11:19:31.707 : N8059 M105*19
11:19:31.707 : ok 8059
11:19:31.707 : T:154.44 / 0 B:81.84 / 0 B@:0 @:0 T0:154.44 / 0 @0:0 T1:48.34 / 0 @1:0
11:19:32.705 : wait
11:19:33.704 : wait
11:19:34.702 : wait
11:19:34.765 : N8060 M105*25
11:19:34.765 : ok 8060
11:19:34.765 : T:152.01 / 0 B:81.67 / 0 B@:0 @:0 T0:152.01 / 0 @0:0 T1:48.12 / 0 @1:0
11:19:35.763 : wait
11:19:36.761 : wait
11:19:37.760 : wait
11:19:37.822 : N8061 M105*24
11:19:37.822 : ok 8061
11:19:37.822 : T:149.61 / 0 B:81.51 / 0 B@:0 @:0 T0:149.61 / 0 @0:0 T1:48.12 / 0 @1:0
11:19:38.821 : wait
11:19:39.819 : wait

```

businka, 2017-07-31:
Repeated again. The problem did not appear.
I do not understand what causes this problem

```
12:23:29.845 : ok 24825
12:23:29.845 : N24832 M105*40
12:23:29.892 : ok 24826
12:23:29.892 : N24833 G1 X76.342 Y59.892 F7800*69
12:23:29.892 : N24834 G1 E2 F2400*22
12:23:30.032 : ok 24827
12:23:30.266 : ok 24828
12:23:30.266 : N24835 M117 ETE/ 1h 49m 59s*17
12:23:30.313 : ok 24829
12:23:30.313 : N24836 G1 X73.663 Y59.892 E2.11214 F3600*7
12:23:30.422 : ok 24830
12:23:30.422 : N24837 G1 X73.704 Y57.035 E2.23175*107
12:23:30.469 : ok 24831
12:23:30.469 : N24838 G1 X76.302 Y57.035 E2.34047*101
12:23:30.469 : ok 24832
12:23:30.485 : T:230.38 /230 B:89.27 /90 B@:255 @:230 T0:230.38 /230 @0:230 T1:49.66 / 0 @1:0
12:23:30.485 : ok 24833
12:23:30.485 : N24839 G1 X76.379 Y54.178 E2.46011*101
12:23:30.641 : ok 24834
12:23:30.641 : N24840 G1 X73.628 Y54.178 E2.57526*110
12:23:30.703 : ok 24835
12:23:30.719 : N24841 G1 E0.57526 F2400*11
12:23:30.719 : N24842 G92 E0*79
12:23:30.719 : ok 24836
12:23:30.719 : N24843 G1 X73.669 Y51.32 F7800*113
12:23:31.624 : ok 24837
12:23:31.624 : N24844 M104 T0 S0*47
12:23:31.624 : N24845 M104 T1 S0*47
12:23:31.670 : ok 24838
12:23:31.670 : N24846 M140 S0*105
12:23:32.138 : ok 24839
12:23:32.138 : N24847 M84*18
12:23:32.201 : ok 24840
12:23:32.201 : N24848 M105*37
12:23:32.606 : ok 24841
12:23:32.669 : ok 24842
12:23:32.669 : ok 24843
12:23:33.589 : ok 24844
12:23:34.634 : N24849 M105*36
12:23:35.602 : busy:processing
12:23:36.616 : TargetExtr 0: 0
12:23:36.616 : Printed filament:206.84m Printing time:1 days 13 hours 25 min
12:23:36.616 : ok 24845
12:23:36.616 : TargetExtr 1: 0
12:23:36.616 : ok 24846
12:23:36.616 : TargetBed: 0
12:23:36.616 : ok 24847
12:23:36.647 : T:228.60 / 0 B:93.17 / 0 B@:0 @:230 T0:228.60 / 0 @0:230 T1:49.88 / 0 @1:0
12:23:36.647 : ok 24848
12:23:36.662 : T:228.60 / 0 B:93.17 / 0 B@:0 @:230 T0:228.60 / 0 @0:230 T1:49.88 / 0 @1:0
12:23:36.662 : ok 24849
12:23:36.662 : T:228.60 / 0 B:93.17 / 0 B@:0 @:230 T0:228.60 / 0 @0:230 T1:49.88 / 0 @1:0
12:23:37.661 : wait
12:23:37.676 : N24850 M105*44
12:23:37.676 : ok 24850
12:23:37.676 : T:228.88 / 0 B:93.02 / 0 B@:0 @:0 T0:228.88 / 0 @0:0 T1:49.88 / 0 @1:0
12:23:38.675 : wait
12:23:39.673 : wait
12:23:40.672 : wait
12:23:40.734 : N24851 M105*45
12:23:40.734 : ok 24851
12:23:40.734 : T:230.04 / 0 B:91.64 / 0 B@:0 @:0 T0:230.04 / 0 @0:0 T1:49.88 / 0 @1:0
12:23:41.732 : wait
12:23:42.731 : wait
12:23:43.729 : wait
12:23:43.807 : N24852 M105*46
[... 193 lines cut ...]
12:25:28.005 : ok 24891
12:25:28.020 : SelectExtruder:0
12:25:28.020 : FlowMultiply:100
12:25:28.129 : N24892 M105*34
12:25:30.003 : busy:processing
12:25:30.253 : X:0.00 Y:0.00 Z:0.000 E:0.0000
12:25:30.253 : ok 24892
12:25:30.269 : T:137.86 / 0 B:82.61 / 0 B@:0 @:0 T0:137.86 / 0 @0:0 T1:47.03 / 0 @1:0
12:25:31.189 : N24893 M105*35
12:25:31.189 : ok 24893
12:25:31.220 : T:137.30 / 0 B:82.57 / 0 B@:0 @:0 T0:137.30 / 0 @0:0 T1:47.03 / 0 @1:0
12:25:32.187 : wait
12:25:33.186 : wait
12:25:34.184 : wait
12:25:34.247 : N24894 M105*36
12:25:34.247 : ok 24894
12:25:34.247 : T:135.40 / 0 B:82.41 / 0 B@:0 @:0 T0:135.40 / 0 @0:0 T1:46.81 / 0 @1:0
12:25:35.245 : wait
12:25:36.243 : wait
12:25:37.148 : N24895 G1 Z5*101
12:25:37.148 : ok 24895
12:25:37.304 : N24896 M105*38
12:25:37.304 : ok 24896

```

luc-github, 2017-07-31:
looking at the code I found the "x3" 
```
bool Printer::isPositionAllowed(float x,float y,float z)
{
    if(isNoDestinationCheck()) return true;
    bool allowed = true;
#if DRIVE_SYSTEM == DELTA
	if(!isHoming()) {
	    allowed = allowed && (z >= 0) && (z <= zLength + 0.05 + ENDSTOP_Z_BACK_ON_HOME);
		allowed = allowed && (x * x + y * y <= deltaMaxRadiusSquared);
	}
#else	// DRIVE_SYSTEM
	if(!isHoming()) {
		allowed = allowed && x >= xMin;
		allowed = allowed && x <= xMin + xLength;
		allowed = allowed && y >= yMin;
		allowed = allowed && y <= yMin + yLength;
		allowed = allowed && z >= zMin;
		allowed = allowed && z <= zMin + zLength + ENDSTOP_Z_BACK_ON_HOME;	
	}
#endif 
#if DUAL_X_AXIS
	// Prevent carriage hit by disallowing moves inside other parking direction.
	if(Extruder::current->id == 0) {
		if(x > xMin + xLength)
			allowed = false;
	} else {
		if(x < xMin)
			allowed = false;
	}
#endif
    if(!allowed)
    {
	Com::printFLN(PSTR("x3"));
        Printer::updateCurrentPosition(true);
        Commands::printCurrentPosition(PSTR("isPositionAllowed "));
    }
    return allowed;
```
but I do not see anything wrong in code - it just means position is not allowed because out of range

luc-github, 2017-07-31:
Ok reading / comparing code/reviewing history - it seems x3 come from me when I was debuging something when implementing 0.92.10 but no idea what

Seems also this function has been updated on mainstream - currently we are little bit behind repetier development as all changes are related to different printer / system, but sometime they also may affect /fix some random issues

So I have updated 2 functions hoping it solve the current issue : https://github.com/luc-github/Repetier-Firmware-0.92/commit/832ddaf6cf79bbfdc11516656906bdcab2285a02

Please have a try as I cannot reproduce your issue to see if it fixes it

Thanks

businka, 2017-08-01:
thanks for the help. When do you plan to release the next version? I do not want to solve this problem to get many others. I learned to bypass the problem.

luc-github, 2017-08-01:
I have only updated the functions I mentioned - I do not know when next release will happen, before end of year I think

luc-github, 2018-02-03:
I close issue feel free to open again if needed

closed 2018-02-03 by luc-github

megejas, 2023-10-05:
For me solution to this problem was to disable autoleveling :) No more x3 faults