luc-github/Repetier-Firmware-4-Davinci #258: problem

demotte9, 2017-07-02:
Hello, i made all the step to flash my da vinci. The last command on my arduino board is CPU reset. Anyway, there is always the two black lines on the printer and my computer doensn't reconize the printer now... I try 3 times but always the same problem.. Please help !

[image]

[image]

luc-github, 2017-07-02:
it is typical issue : https://github.com/luc-github/Repetier-Firmware-0.92/issues/172

luc-github, 2017-07-16:
Ok I close issue if no update

closed 2017-07-16 by luc-github

demotte9, 2017-07-17:
yes, i try this weekend but it doesn't work, i will make some update with
print screen this week.

Regards,

Antoine