luc-github/Repetier-Firmware-4-Davinci #259: Unhandled communication error... Error: Could not write eeprom to sd card - no sd card mounted

ZeekWorld, 2017-07-07:
Hi,
I've been running an older version of the 0.92 for years without issues, a week ago I upgraded the extruder with a v6 and flashed the latest 0.92.10 , I've been getting the following error: "Unhandled communication error... Error: Could not write eeprom to sd card - no sd card mounted" It happens at random times and when happening during a print job, the printer stops responding. I am not sure what this is about. Any help would be appreciated it.
I can also provide more information if needed. I have attached a screenshot. Thank you in advance!﻿
https://www.dropbox.com/s/0bmke4qympjp14u/Image%20319.jpg?dl=0

The error message pops up when running octoprint / octopi. I contact the octoprint community and they told me is should have something to do with the repetier firmware which make sense since I haven't done anything with octoprint but I did flashed rhe latest version of Repetier-Firmware-0.92 on my Da Vinci.

Here is the response from the Octoprint community https://plus.google.com/+ZeekFerrarós/posts/U8U88GXekAu

ZeekWorld, 2017-07-07:
Just wanted to add, I found a pattern, I get the error at the end of every print.
This is what shows up on the Terminal,

Recv: Info:Case lights off
Recv: wait
Send: N29998 M105*36
Recv: ok 29998
Recv: T:208.28 / 0 B:40.97 / 0 B@:0 @:0
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Send: N29999 M105*37
Recv: ok 29999
Recv: T:201.78 / 0 B:40.84 / 0 B@:0 @:0
Recv: Error:Could not write eeprom to sd card - no sd card mounted
Changing monitoring state from 'Operational' to 'Error: Could not write eeprom to sd card - no sd card mounted\x0d\x0a'
Recv: wait
Recv: wait

luc-github, 2017-07-07:
I never saw this error - did you tried to revert previous version to confirm issue follow firmware and it is not a bad luck ?

What was the working version ?

Can you share your end GCODE ? if it is only at the end of the print job and not during print, it means some command is messing up with FW, so should be part of end GCODE

ZeekWorld, 2017-07-07:
The code I've attached, I've use it probably 20 times with the previous 0.92 firmware without issues.
To answer your question about the working version, I never paid attention to the build and now it's gone since I've upgraded to 0.92.10, what I can tell you I installed 0.92 around May 2015 (whatever 0.92 build was around that time is what I've been running without issues until last week's flash upgrade).
To add some more info, the issue happened with older GCODEs I have saved and new GCODES I downloaded recently and sliced them.
GCODE (this the the one I have saved and used multiple times in the past and gave me the error code at the end of the print with the new firmware) https://www.dropbox.com/s/68ynly1gj736jcl/Solo_Legs_Short_x5_Center_S.gcode?dl=0 

Thanks!

luc-github, 2017-07-07:
I do not see anything  wrong in end GCODE but anything could be wrongly interpreted by FW even some comment.

to narrow down and eliminate hw issue can you :
1 -  try print same code but remove all comments at the end and remove the light off command, that may try to update EEPROM when it should not
2 - reverte to previous versio https://github.com/luc-github/Repetier-Firmware-0.92/releases/tag/v0.92.3.0

One difference between 92.3 and 0.92.10 is : virtual EEPROM is now managed by repetier code and writing update cycle are different

ZeekWorld, 2017-07-07:
Thank you for the quick response, 
1 - I just removed the end code and I am printing a test to see how it goes.
2 - I would definitely try 92.3 and report back.

Zeek

ZeekWorld, 2017-07-08:
Report:

1- Removing the end code didn't make a different.
2- Flashed 92.3 and the issue is gone!!
Now, if the issue is with the EEPROM being managed by repetier, is there anything I can do to run the latest firmware 92.10 or I shouldn't be concerned of keeping 92.3.
Just wondering if I am missing something by running an older firmware.
Thanks a lot!!

luc-github, 2017-07-08:
0.92.10 has several bug fixes including a working autolevel according to repetier and support multilanguage at runtime

as I am in moving process, and my printer is packed , it is hard to work on reproduce issue now - but looking at code : https://github.com/luc-github/Repetier-Firmware-0.92/blob/7f28c5eded5f35210429f42aaba72ec0e9796036/src/ArduinoDUE/Repetier/HAL.cpp#L247-L254

here some test you can do :
1 - try to print from SD instead of host
2 - if issue happen, without reseting anything check if printer can list  SDfiles from printer menu using print files, and also if printer answer properly with M20 command from Host
3 - if nothing works in 2 send M21 command to init SD and retry 2

this will help to see if somehow SD card is released or if SD card is loose and removed

what is your printer model ? it use standard SD or Micro SD ?

ZeekWorld, 2017-07-08:
I have a Da Vinci 1.0a, I thought I couldn't use the SD, since it's glue to the motherboard. And I am using Octopi on a Raspberry Pi.

luc-github, 2017-07-09:
glue can be removed - several people did it 
if you cannot print from SD then print as usual and send commands I have listed after error happen from octopi web interface

ZeekWorld, 2017-07-14:
Yes, commands don't fix the problem. 
After every print I get the same SD Eeprom message, have to refresh octoprint and reconnect every time on 92.10

luc-github, 2017-07-14:
but what the results of the commands ?

ZeekWorld, 2017-07-14:
oh, sorry, I didn't take the output. I will paste it soon.

ZeekWorld, 2017-07-14:
Just tried to send commands over the terminal after disconnect, not possible. Octoprint terminal won't let me send commands unless I refresh the browser and re-connect by clicking the connect button on the portal.
After getting re-connected I can send the two commands but is useless at this point since I am connect it already.

luc-github, 2017-07-14:
I do not get, the issue is on accessing EEPROM or accessing printer ?

ZeekWorld, 2017-07-14:
When a print is finished, the printer gets disconnected from Octoprint and on the terminal displays the EEPROM message. I also noticed if I am browsing the printer's menu and hit the Home button, it also disconnects from Octoprint displaying the same eeprom sd message.
This behavior does not occur with the older version of Repetier 92.3
Sorry, I am not an expert on coding, I am just trying to explain what has been going on with my printer since I've upgraded to Repetier 92.10.
Maybe this is just a behavior between Repetier 92.10 and Octoprint and I will have to live with it. The one thing I know, the Octoprint developer pointed me to the Repetier firmware  and is why I started the thread over here.

luc-github, 2017-07-14:
disconnecting and error accessing EEprom on SD card are 2 differents problems
I do not think disconnection is an big issue as you can reconnect after print is done,  right ?

also did you applied the USBcore.cpp on 0.92.3 and on 0.92.10 ?

ZeekWorld, 2017-07-14:
I followed the instructions https://github.com/luc-github/Repetier-Firmware-0.92
There is no mentioned of the USBcore.cpp file.
Anyways, I did flashed once without copying the above file and another one with the USBcore.cpp. Same results.

luc-github, 2017-07-15:
hum just checked seems there were some glitch when I merged latest version - readme is broken and miss USBCore.cpp reference - fixed

Back to EEPROM issue - Can you get commands result when reconnecting after issue happen ? you mentioned that you must reconnect but you did not share commands result

luc-github, 2017-07-23:
any update on this ?

ZeekWorld, 2017-07-23:
The one update I have is that I found out another user with similar setup has the exact same problem.

luc-github, 2017-07-23:
I do not said there is no issue - I just cannot reproduce it so no idea how to investigate
as wrote before : 
 Can you get commands result when reconnecting after issue happen ? you mentioned that you must reconnect but you did not share commands result

ZeekWorld, 2017-07-23:
Sure will do.

ZeekWorld, 2017-08-01:
Here you go...

**Started a print, after the print finished, printer disconnects and get the "unhanded communication error" on the Octoprint portal.
Output from terminal below**

Recv: Printed filament:1.59m Printing time:0 days 0 hours 22 min
Recv: ok 20832
Send: N20833 M140 S0*111
Recv: ok 20833
Send: N20834 M84*18
Recv: ok 20834
Send: N20835 M107*41
Recv: ok 20835
Changing monitoring state from 'Printing' to 'Operational'
Recv: Unknown command:N20835 M107
Send: N20836 M105*40
Recv: ok 20836
Recv: T:229.29 / 0 B:29.36 / 0 B@:0 @:0
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Send: N20837 M105*41
Recv: ok 20837
Recv: T:229.01 / 0 B:29.63 / 0 B@:0 @:0
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Send: N20838 M105*38
Recv: ok 20838
Recv: T:224.22 / 0 B:29.36 / 0 B@:0 @:0
Recv: wait
Recv: wait
Recv: wait
Recv: wait
**Recv: Error:Could not write eeprom to sd card - no sd card mounted
Changing monitoring state from 'Operational' to 'Error: Could not write eeprom to sd card - no sd card mounted**
'
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait

**I have no choice but to manually reconnect, on Octoprint I have to click on "Connect" under Connection
Output from terminal below**

Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Changing monitoring state from 'Error: Could not write eeprom to sd card - no sd card mounted
' to 'Offline'
Connection closed, closing down monitor
Changing monitoring state from 'Offline' to 'Detecting serial port'
Serial port list: ['/dev/ttyACM0']
Connecting to: /dev/ttyACM0
Changing monitoring state from 'Detecting serial port' to 'Opening serial port'
Connected to: Serial<id=0x6d165430, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Starting baud rate detection
Changing monitoring state from 'Opening serial port' to 'Detecting baudrate'
Trying baudrate: 115200
Send: N0 M110 N0*125
Recv: ok
Changing monitoring state from 'Detecting baudrate' to 'Operational'
Send: N0 M110 N0*125
Recv: ok
Send: N1 M115*39
Recv: ok 1
Send: N2 M105*37
Recv: FIRMWARE_NAME:Repetier_0.92.10 FIRMWARE_URL:https://github.com/luc-github/Repetier-Firmware-0.92/ PROTOCOL_VERSION:1.0 MACHINE_TYPE:DaVinci EXTRUDER_COUNT:1 REPETIER_PROTOCOL:3
Recv: Printed filament:1.59m Printing time:0 days 0 hours 22 min
Recv: PrinterMode:FFF
Recv: ok 2
Recv: T:83.34 / 0 B:29.36 / 0 B@:0 @:0
Recv: wait
Recv: wait
Recv: wait
Recv: wait

luc-github, 2017-08-01:
What is the baud rate in lcd printer settings ?

ZeekWorld, 2017-08-02:
230400

luc-github, 2017-08-02:
your printer is set for 230400 but your host use 115400
>Connected to: Serial<id=0x6d165430, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, 

this can bring unexpected behaviour

ZeekWorld, 2017-08-03:
Baudrate was set to "Auto", after hardcoding it to 230400, unfortunately didn't make any difference. Still dropping the connection with printer and server at 230400.

ZeekWorld, 2017-08-09:
After numerous print failures because of disconnects while printing I've decided to downgrade and flash 092.3.
Since then, the printer has been running solid on the older firmware when connected to octoprint.
My conclusion after lots of testing is that Davinci 1.0a and latest 92.10 when using Octoprint/Octopi will cause print failures and can even be dangerous to leave jobs unattended. In my case, every time the printer lost connection with the server while printing, the extruder stopped moving but continued to heat, same for the bed, this started melting the print causing a dangerous mess.
I would recommend users to stay on 92.3 if running octoprint.

luc-github, 2017-08-09:
I cannot reproduce any of the problem sorry
I am syncing latest repetier code may be this will solve these issues -  at least the 0.92.3 is working for you

luc-github, 2018-02-03:
I close issue as no one else reported it and will sync to 1.x in close future

closed 2018-02-03 by luc-github

nmielec, 2026-10-11:
Hello, 6 years later, I got an AIO 1.0 and flashed the firmware, and hit same issue.

I removed the original SD Card and put another one, formatted FAT32, and the bug was solved.

I checked the original SD Card and it was not formatted, showed as "RAW" in Windows Disk Manager.
I formatted it to FAT32, inserted it, and 0.92.10 works without issues.