luc-github/Repetier-Firmware-4-Davinci #260: Unload filament

TComp, 2017-07-15:
I have been using my DaVinci Duo for a few years with Repetier v0.91 without any problem (thanks a lot, by the way). I flashed it today with 0.92.10 and everything seems to work fine except for 2 issues: filament can't be unloaded and Y axis tries to move past origin. After the left extruder reaches 230C, "Press OK to continue" appears for a fraction of a second and the carriage moves to the origin. The right extruder exhibits the same behaviour, but the carriage moves to the left about 50-60 mm.

Am I doing something wrong?

Thanks

luc-github, 2017-07-16:
Did you load fail safe settings after flashing ?
in 0.92 I have put load/unload settings like distance and acceleration in EEPROM for easy adjustment so if you did not do the value may be to 0
Check the values using repetier EEPROM editor

TComp, 2017-07-16:
I did load fail-safe. The loading filament function works fine, just unloading won't work. Flashed 0.92.3 back again and works flawlessly, re-flashed with 0.9.10 and the same problem appears again. Now I'm back to 0.92.3.
Right now cannot run any tests since I am replacing the stock extruders with new ones, but will try again when done later this week. Thanks for the editor suggestion; will install the plugin in Octoprint and check the values.

luc-github, 2017-07-16:
I should have my duo unpacked by end of this week - I will see if I can reproduce - it is first generation as yours but with only one fan so I should be able to reproduce if it is not a setting issue

TComp, 2017-07-17:
Thank you. Once I finish my hotends project, will reflash 0.9.10 and will check the acceleration and distance settings. My Duo has two fans but I don't think that will make any difference.

luc-github, 2017-07-23:
I just verified now my duo is unpacked and cannot reproduce your issue which should lead to setting issue in EEPROM

TComp, 2017-07-23:
Thanks for testing the issue. I will be able to re-flash with the new version in a couple of days and check again. Will check the EEPROM parameters, too.

luc-github, 2017-07-23:
Ok thanks

TComp, 2017-07-30:
Sorry, I haven't finished modding my printer yet. I replaced the stock extruder with two E3D V6 clones, changed from direct to Bowden feeding and now I am replacing the Z-axis threaded rod with a lead screw. Even if I flash the firmware, won't be able to test it since this work is still in progress. I am planning to finish it on Monday and then flash the firmware. Will let you know as soon as I'm done.

TComp, 2017-07-31:
I was able to test it again today and it is having the same issue. As soon as the extruder temperature reaches 230C, the extruder turns off and the carriage moves to home position without unloading the filament. Checked the unloading parameters with the EEPROM editor and everything looks OK.: unload feed rate 4, unload/load distance 60, acceleration for X and Y axis 1000. Switched back to 0.92.3 and everything works fine.

luc-github, 2017-07-31:
is filament sensor setting disabled ?

TComp, 2017-07-31:
Will have to flash again and come back to you. In 0.92.3 the filament sensor is on and unloading works fine.

TComp, 2017-07-31:
It appears that that was the problem. The filament sensor was ON; when I turned it off, unloading worked perfectly. On the other hand, in 0.92.3 unloading works even if that setting is ON. Thanks a lot for the support.

luc-github, 2017-07-31:
o'k sometbing should be wrong with filament sensor
does it happen  on both extruders  ?

TComp, 2017-07-31:
The right extruder behaves similarly: when the temperature reaches 230C, the extruder turns off and the carriage moves to the left about 50-60 mm and stops. When you say something wrong with the sensor means a hardware issue or some setting?

luc-github, 2017-07-31:
my sensor is always off so I may missed something when testing
I was talking about code as 0.92.3  is fine

TComp, 2017-07-31:
You may want to try on your printer and verify if you can replicate the problem. How about the top cover sensor? I turned that off, too along with the filament sensor.

luc-github, 2017-07-31:
top sensor just generate sound if heating  extrudernand top is open
 yes will check on my printer tomorrow to replicate with sensor on

luc-github, 2017-08-01:
Ok I can reproduce issue now -  will see what is wrong with sensor code

luc-github, 2017-08-01:
I have review all the code and same code was working before so without a clue about root cause I have done some code cleaning and add sensor report status function, but during my investigation I got some issue reading proper sensor status - I am not sure it due to hardware failure or it was due to some memory management issue, 
I have modified variant.cpp to not have any pin over 127 to avoid overflow as value must be -127/+127 to fit variables.
I have also get back from 1.8.3 to 1.8.0 as even on 0.92.3 I was not able to see the sensors status with consistency.
I have fixed couple of issues with the  periodical door sensor check  and added the M121 command to get sensor status report

Now I cannot reproduce the issue anymore and the sensor status is now easy to see when loading unloading

please have a look and feedback if also solve issue on your side

TComp, 2017-08-03:
I haven't been able to test the updated firmware yet since the prniter has been running all time. I hope to be able to test it this weekend. Meanwhile, I noticed some other issues with the firmware. I have created some custom controls in Octoprint: turn light on/off, lower bed, disable steppers, lower bed, etc. None of them is working now with the 0.92.10 while they have been working fine with 0.92.3. Not sure if I am doing something wrong but I haven't changed anything except for the firmware.

luc-github, 2017-08-03:
what is your baud rate in lcd printer settings ?
what is your baud rate in octopi

TComp, 2017-08-03:
230400 on both. Octopi.log is filled up with warnings: printer requested line 216543 but no sufficient history is available, can't resend. I am noticing now that none of the commands in the control tab is working, not only my custom controls. I cannot move the axes or anything else in the page. On the other hand, I can upload files, start, pause or cancel prints and watch the webcam stream. Also, sending a manual command from the terminal tab does not work.

luc-github, 2017-08-03:
I do not use octopi ,  I only print from SD so I cannot comment for octopi issue
But if print is working and some commands works - I would say the issue is not the FW 
I have connected repetier host and the manual tab is working well all command answers
same behavior if I use my wifi web UI which just send GCODE separated by newlines

I suggest you to check what is the log when you send command from repetier host and what is the log when send from octopi

TComp, 2017-08-03:
Well, while I was testing, it started working again out of the blue. It is possible that my computer had some kind of problem, not sure. Anyway, I tested your new code and unloading now works with the sensor on, so problem solved. I noticed only a couple of different things: when loading or unloading, on the LCD appears the message "Push OK to start" but it does not wait for it and loading/unloading starts automatically. Also, when the top cover sensor is on, loading/unloading stops for a moment when the printer is beeping and the restarts again. Not a problem for me, just thought you would have liked to know. Thanks again for your great code. With the this firmware and some hardware mods, I am getting high quality prints from this machine with a lot of different materials.

luc-github, 2017-08-03:
Good catch - let me answer:
about message: the sensor is  kind of automation, so message start to push may not be the best but it is displayed in case of no sensor or if filament is transparent to launch the extrusion, if sensor is on and filament is detected the message disapear and extrusion start automaticaly

about beep : well this is limitation of FW as it is not multithread so when beeping no other action is done and the motor may wait the end of beep to continue - I agree this should be disabled when loading / unloading as top cover need to be open - I will add this is todo list

About mod:
* I would sugest to use Wifi SD card to print from SD - very convenient to upload files as Serial upload is a disaster
* if you want to monitor SD print I also suggest to add Wifi interface using ESP3D   

Happy printing

luc-github, 2017-08-03:
I have opened an issue for Disabling door sensor when loading/unloading
So as current issue is solved I close it

closed 2017-08-03 by luc-github

TComp, 2017-08-03:
Yes, as far as I am concerned, the issue has been resolved. Thank you again.