luc-github/Repetier-Firmware-4-Davinci #261: Fakes printing super fast

bj97301, 2017-07-22:
First, thank you for the amazing work on this firmware.  

The issue I am having is that when I want to print a gcode file it seems like its going to work but after the start scripts are done the actual print commands dont seem to do anything.  The printer acts likes its printing super fast from what I can see from the terminal but the motors are not moving at all.  I have tried generating the gcode file in repetier host mac 1.0.2, s3d and mattercontrol.  I have also tried octoprint and hooking it up to my mac with all of the above software and they all seem to do the same thing. 

Thank you in advance for your help.

bj97301, 2017-07-23:
Strange... If I take out the G28 command at the end of my start script it seems to recognize the G1 commands but the bed is too low.  Going to play around with a bit to see if I can figure out whats going on.  Below are my start scripts:
G28 ; home all
M100 ; clean shitz
G28 ; home all again

luc-github, 2017-07-23:
can you share your start script ?

luc-github, 2017-08-09:
I close issue if no update

closed 2017-08-09 by luc-github