luc-github/Repetier-Firmware-4-Davinci #262: No USB Detection

IM6UR9, 2017-07-24:
Hi, I have the Davinci 2.0a duo after trying a couple of times to upgrade FW to Repetier, the printer now will not register at all in my usb port(s), any ideas how to make the printer be recognized.
I have searched but nothing seems to work, any help would be greatly appreciated.
Thanks

luc-github, 2017-07-24:
your printer works or there are black bare on lcd ?

_Sent from my Lge LG-H990 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

IM6UR9, 2017-07-24:
I have jumped the j37 so only black bars on lcd, when I power it on the fan works briefly as it dose in this state, also the bed heats up very hot so I can't leave it on for long.

luc-github, 2017-07-24:
well in that case , if after using jumper printer is not properly detected then issue is on computer side, try restarting computer or reinstall arduino drivers

_Sent from my Lge LG-H990 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

IM6UR9, 2017-07-24:
From what I gather its not the computer.
 I have tried to install the arduino drivers (I think i tried it right),in the device manager but I don't get any recognition of the device when plugged in, can't even find  (COM & LPT) and "Other Devices" for "Unknown Device". when printer is plugged in.
If I plug  another device in (unrelated to this printer) the port(s) work fine.
The printer was getting recognized previously while trying to do the upgrade but I mucked a few things up (my fault) on my first couple of attempts. so I started all over again and now the USB is not recognizing the printer.

luc-github, 2017-07-24:
what did you do wrong ?

if no chime neither unknow device when plugged so yes can be on printer side, but using JP 37 wipe FW, so it means you did a short during your tries ?

_Sent from my Lge LG-H990 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

IM6UR9, 2017-07-24:
No chime ether, just nothing.
Very sure I didn't short or touch any other components in the back of the printer. This is what has me stumped, I did short that Jp37 3 or so times, in trying to do the FW upgrade, I did this every time I couldn't get it to upgrade the FW, I started so called fresh every time, uninstalled Arduino and Repetier and downloaded again, I think I have it all right now ready to go in that side of things, but now no printer recognition in USB to try it.

luc-github, 2017-07-24:
humm, I can only say try to wipe, keeping the jumper connected , power on, wait few seconds, power off, power on again wait few sec then remove jumper, power off, wait,  then power on

_Sent from my Lge LG-H990 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

IM6UR9, 2017-07-24:
Righto will give that ago a see what happens, and thanks for your time!!
Will let you know

IM6UR9, 2017-07-24:
No luck.
Mmmm......time to buy a new printer ha ha
I was so close to getting it updated to!!!  
Have you heard of this problem before? or do you know of any other way to give life back to the printers usb port?

luc-github, 2017-07-24:
no, as it is normal way to wipe and flash this MCU, and the wipe work at hardware level

one people got issue on first generation and asked for board change but it was bad luck as he was able to flash successively several time before

do you have the 3 leds ok on the board?

_Sent from my Lge LG-H990 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

IM6UR9, 2017-07-24:
Funny you mention that I just noticed only two, I just uploaded the image of the two lights, hope it works!
[image]

luc-github, 2017-07-24:
well looks like there was a short and some fuse blowed up, hopefuly it is not difficult to fix from what I saw on voltivo and soliforum forums.
I cannot link now as I am in train and using phone. will do later if cannot find

_Sent from my Lge LG-H990 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

IM6UR9, 2017-07-24:
Ha, ok thanks, looks like the 3.3volt light inop.
I will have a dig around on those two forums.

luc-github, 2017-07-30:
can we close issue as it is Hardware issue

IM6UR9, 2017-07-30:
Sure, and thanks for your help.

closed 2017-07-31 by luc-github

TComp, 2017-08-03:
I had a problem like this when I shorted the 5V and 12V pins. It fried the chip and the printer wasn't recognized at all. I tried to find another board (I have a Duo) on Ebay or Amazon without luck. There are a couple of solutions here depending how good you are with soldering. You can just install a completely different board (RAMPS or equivalent) or replace the chip. I ended up buying a new ATMEL chip at Quest Components for $12 (or you can get elsewhere, like Mouser Electronics), cut the pins of the old chip from the board, clean the connections and solder the new chip. Must say this is not something for the faint of heart; it took me 4-5 hours to cut away the old one with a very sharp surgical tool and solder the new one. All the work was done under microscope; I do not believe there is any other way to do this with a 3/4" size chip with a total of 144 pins. You have to solder and check all pins one by one making sure they are soldered correctly and do not bridge with each other. Good luck!

IM6UR9, 2017-08-11:
Well finally got around to investigating and found the problem, thought I would share the fix, the main board had a blown surface mounted fuse, it was R57 in the 3volt circuit. right near the black input power connector. 
Replaced and tested for USB on PC and it all worked, flashed the firmware and it's all working!! :)

Now with repiter host, it all works,  but it always says 'no filament' in extruder. but there is
Any ideas?

luc-github, 2017-08-11:
Ok
you can disable sensor check in settings
if it says no filament there a 3 possible reasons 
* as you are using duo you heat both extruders when you may be use only 1 and the one you do not use does not have filament
* you use clear filament which is not detected
* your daughter board is disconnected or connector on mainboard for sensor is disconnected, when no connection the sensor status id no filament by default, so you beed to find where is the connection issue

IM6UR9, 2017-08-11:
Ok thanks got it to work, but now filament isn't coming out of the nozzle,
its moving around like it should be but just not extruding, printing
nothing.

luc-github, 2017-08-11:
does load filament feature works from printer ?
what is the temperature when trying to extrude during printing?

_Sent from my Lge LG-H990 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

IM6UR9, 2017-08-11:
Yes load filament works fine @ a temp of 200.
Also the extrude temp while trying to print is @ 200. Using PLA

luc-github, 2017-08-11:
if extrusion is Ok with load filament - it is not Fw issue
can be : 
* slicer setting issue
* bed leveling issue : bed is too close not allowing proper extrusion
* you use PLA which is not XYZ PLA, and your extruder is jammed, stock extruder does not work well with third party PLA filament, there are lots of tricks to do it but quality is not great
  check my very old post : https://forum.voltivo.com/showthread.php?tid=8283&highlight=PLA

To print third party PLA on Davinci it is recommended to change extruder

IM6UR9, 2017-08-11:
I've been using 3rd party filament for around 9 months now with no problem
on stock FW, (except that I can't change to the precise temp on the
extru/bed for a particular filament.)
The bed is set up right, not stopping flow
What's the best slicer to use with this FW?
I think it maybe settings, is there any guide to set up the slicer for the
2.0a?
Have been trying to use repiter host with no luck.

JMcGraw47, 2020-01-19:
Same thing happened to me last night trying to flash Repetier to 2.0A Duo. First attempt display lit up but still had bars, second attempt just bars and no back light, 3rd try USB error.
    Looking at maybe a diff board?

luc-github, 2020-01-19:
No USB detection on window  is usually due to variant not properly set, check FAQ have all infos

JMcGraw47, 2020-01-19:
Thank you, trying to load Repetier to xyz 2.0A Duo and the usb error is
random.

John McGRaw

luc-github, 2020-01-19:
usb detection issue or black bar have same root cause. if FW is loaded and printer is ok but has usb detection then it is another issue than variant no set properly