luc-github/Repetier-Firmware-4-Davinci #263: Sync with mainstream

luc-github, 2017-07-31:
Long this has not be done -  may time to do it

luc-github, 2017-08-08:
I am currently syncing with current devt branch of repetier to get latest fixes
Doing some code simplification, I am wondering if it is necessary to keep the easy/ advanced modes or to stay with only one complete mode like repetier does ?
 
So this would lead to use original repetier UI menu + specific menus of Davinci dispatched on different repetier submenus  and no more currrent fully customized one 

Any feedback on this ?

cdoc83, 2017-08-14:
I vote for going with original repetier menu + additional DaVinci specific menus.

rhinorap, 2018-04-16:
Seconded, but keep in mind that your menu structure is bang on with the current build and the repetier menus are somewhat.......clunky.

luc-github, 2018-04-16:
The current menu for davinci is rewrited from repetier with feature not present in repetier and based on early discussion of usage which are little bit different than repetier spirit
This bring 80% of the work to maintain it because not really easy to do code difference on very different source code
This include repetier wizard mode and davinci wizard mode which are not compatibles...

That is why I was wondering if it worth to maintain Davinci UI. As I never got users feedback on it

luc-github, 2018-06-02:
Now duplicate with https://github.com/luc-github/Repetier-Firmware-4-Davinci/issues/286
closing

closed 2018-06-02 by luc-github