luc-github/Repetier-Firmware-4-Davinci #265: Resume printer after M600

patator03, 2017-08-03:
Hi , i have some troubles with filament change during printing .

When i put M600 , printer is going to load / unload filament well .

But it makes printer  //;pause and i can't do anything after that , in order to resume and restart the print .

Any solutions for it ?

Thanks

luc-github, 2017-08-04:
What firmware based version are you using ? 0.92.3 or 0.92.10 ?

patator03, 2017-08-04:
hi , i use 0.92.1

luc-github, 2017-08-04:
I guess you mean  0.92.10
looking at the code it seems the loading / unloading is missing the end unlike the previous version.
I originaly planned to let the repetier wizard but it is not working well on Davinci so I have re-introduced the davinc load unload but it seems the end of process is not implemented 

Are you printing from host or from SD ?

patator03, 2017-08-07:
Thanks for your reply

Im always printing with my computer

luc-github, 2017-08-07:
So until it is fixed you may workaround using pause according your Host
for repetier it is `@pause change filament`
for s3d it `M0`
I do not know for others host but every should have

MasterCATZ, 2020-08-01:
as you closed mine 
I flashed my davinci 2.0 with current master 

same issues ,while using repeteir host / repetier server
#define DAVINCI 3// "0" if not DAVINCI, "1" For DAVINCI 1.0, "2" For DAVINCI 2.0 with 1 FAN, "3" For DAVINCI 2.0 with 2 FAN, 4 for AiO (WITH NO SCANNER SUPPORT)
#define MODEL  1//"0" for first generation (jumper JP1 to reset ) , "1" for new generation   (jumper J37 to reset)
#define REPURPOSE_FAN_TO_COOL_EXTRUSIONS 1 //Setting this to 1 will repurpose the main Extruder cooling fan to be controlled VIA M106/M107

if I have fan set as 0 repetier complains about unkown commands set as 1 it does not but still no fan controll how ever someone said leave 0 and change

This further down:

#else
//  #define FAN_PIN -1
//  #define FEATURE_FAN_CONTROL 0
  #define FAN_PIN ORIG_FAN_PIN
  #define FEATURE_FAN_CONTROL 1
#endif

so will try that after this next print attempt 
 

I can only print with extruder #1 (left hand side) , if I select extruder #2 (right hand side) it trys to print with extruder #1 
(I say try because #1 is not heated running fan and #1 is following the print path but #2 is heated no fan running dripping out filament )

also fan control is not really working #1 fan always stays on , but #2 fan seems to be switching on /off with manual settings .. but can not get it to print to #2 extuder at all or control #1 fan ) 

after a print is complete bed  / head stay were they are , if I go maintaince / lower bed the bed tries going up into the hot print head  mushing up the print then comes down 

pretty much the same experiance as a few years ago when I last gave up 

also table is not aligned correctly I think it thinks its 200x 200 when its 150x 200 , even tho I have 150x 200 set 

would be nice to get some use out of this $1000 paper weight  , I had to flash reperiter onto it because I use linux and xyz mostly only supported windows ... I did have this working at one stage with Simplify3D but my licences keep dispearing and given up repurchasing so bought repetier server I still had some original filament in the cartridges thats how much I had used this in the last 6 odd years  only a few sucsessfull prints

luc-github, 2020-08-01:
the stock davinci does not have control to fan by default because it does not have fan to control, you need to add extra fan to use M106/M107 

the issue I closed was related to v1 implementation and was answered, hook an unrelated topic is not good to keep issue clear
please open an issue describing problem if you want some support

and what you describe looks slicer issue, did you checked what gcode is generated ?

closed 2025-11-15 by luc-github