luc-github/Repetier-Firmware-4-Davinci #266: 3D Printer has stopped running the desired G-Code

NexusDevs, 2017-08-09:
Just recently, my 3D Printer has oddly stopped executing the desired G-Code, despite, according to the Repetier-Host, is printing.
On the Host, it surprisingly goes through the print, except a lot faster than how you normally would print.
While on the printer, nothing is happening.
This is all it says at the start of "print"... problem is nothing happens in the printer itself. The commands just start running.

< 4:20:43 PM: Unknown command:N137 M107 
< 4:20:43 PM: TargetExtr 0:200
< 4:20:43 PM: SelectExtruder:0
< 4:20:48 PM: X:-33.00 Y:-12.00 Z:-0.000 E:0.0000
< 4:20:48 PM: x3
< 4:20:48 PM: X:-33.00 Y:-12.00 Z:-0.000 E:0.0000
< 4:20:49 PM: TargetExtr 0:200
< 4:21:55 PM: x3
< 4:21:55 PM: X:-33.00 Y:-12.00 Z:-0.000 E:0.0000
< 4:21:55 PM: x3
< 4:21:55 PM: X:-33.00 Y:-12.00 Z:-0.000 E:0.0000
  4:21:55 PM: Printing layer 1 of 40
< 4:21:55 PM: x3
< 4:21:55 PM: X:-33.00 Y:-12.00 Z:-0.000 E:0.0000
< 4:21:55 PM: x3
< 4:21:55 PM: X:-33.00 Y:-12.00 Z:-0.000 E:0.0000
< 4:21:55 PM: x3
< 4:21:55 PM: X:-33.00 Y:-12.00 Z:-0.000 E:2.0000
< 4:21:55 PM: x3
< 4:21:55 PM: X:-33.00 Y:-12.00 Z:-0.000 E:2.0303
< 4:21:55 PM: x3
< 4:21:55 PM: X:-33.00 Y:-12.00 Z:-0.000 E:2.0808
< 4:21:55 PM: x3
< 4:21:55 PM: X:-33.00 Y:-12.00 Z:-0.000 E:2.1313
< 4:21:55 PM: x3
< 4:21:55 PM: X:-33.00 Y:-12.00 Z:-0.000 E:2.1616
< 4:21:55 PM: x3
< 4:21:55 PM: X:-33.00 Y:-12.00 Z:-0.000 E:2.2323

This is the end of the codes...

< 4:22:40 PM: X:-33.00 Y:-12.00 Z:-0.000 E:176.8586
< 4:22:40 PM: x3
< 4:22:40 PM: X:-33.00 Y:-12.00 Z:-0.000 E:176.8788
< 4:22:40 PM: x3
< 4:22:40 PM: X:-33.00 Y:-12.00 Z:-0.000 E:176.8990
< 4:22:40 PM: x3
< 4:22:40 PM: X:-33.00 Y:-12.00 Z:-0.000 E:176.9293
< 4:22:40 PM: x3
< 4:22:40 PM: X:-33.00 Y:-12.00 Z:-0.000 E:176.9495
< 4:22:40 PM: x3
< 4:22:40 PM: X:-33.00 Y:-12.00 Z:-0.000 E:176.9798
< 4:22:40 PM: x3
< 4:22:40 PM: X:-33.00 Y:-12.00 Z:-0.000 E:177.0101
  4:22:40 PM: Printjob finished at Aug 7, 2017, 4:22:40 PM CDT
  4:22:40 PM: Printing time: 01m:56s
  4:22:40 PM: lines send: 9755
< 4:22:40 PM: x3
< 4:22:40 PM: X:-33.00 Y:-12.00 Z:-0.000 E:177.0404
< 4:22:40 PM: x3
< 4:22:40 PM: X:-33.00 Y:-12.00 Z:-0.000 E:177.0505
< 4:22:40 PM: Unknown command:N9933 M107 
< 4:22:40 PM: TargetExtr 0: 0
< 4:22:40 PM: SelectExtruder:0
< 4:22:40 PM: X:-33.00 Y:-12.00 Z:-0.000 E:0.0000
< 4:22:40 PM: TargetExtr 0: 0
< 4:22:40 PM: x3
< 4:22:40 PM: X:-33.00 Y:-12.00 Z:-0.000 E:0.0000

I cannot see any obvious error on any line.

Your thoughts are greatly appreciated.

Thanks.

luc-github, 2017-08-09:
looks like communication issue
what version of FW are you using ?
Do you enable ping pong communication in repetier host to prevent just send command without ack ?

NexusDevs, 2017-08-09:
I'm using Version 0.92.2.
I tried using Ping Pong communication, which didn't help.
I recorded what happens as a .GIF file to allow you to see what happens in the Host.

[image]

Thank you.

NexusDevs, 2017-08-11:
I've fixed the problem by Reinstalling the Repetier Firmware. 
Thanks for your thoughts and help!

luc-github, 2017-08-11:
Weird - looks like FW corruption in that case - but as FW does not write in flash but SD - it means flash is corrupted somehow, power regulation issue ?  -  be sure your printer is behind an UPS or power protected or it can get worse I think

luc-github, 2017-08-11:
If issue is solved I close it 
Happy printing

closed 2017-08-11 by luc-github