luc-github/Repetier-Firmware-4-Davinci #267: Bed too high

zipippino, 2017-08-28:
Hi, i have a DaVinci 1.0A
I have succesfully installed the repetier firmware but while printing here the error comes. The extruder is too much close to the bed and so the extruder crushes on the printed plastic and makes a disaster. That's because the bed is too much elevated, i think. How could i solve?

[image]
[image]
[image]

luc-github, 2017-08-28:
you must do bed leveling after flashing - did you do it ?

zipippino, 2017-08-28:
Auto leveling directly from the printer? Done, but still nothing

Sent by iPhone   
Francesco Buccoliero aka zipippino
Inforge.net Staff & Graphic Designer

luc-github, 2017-08-28:
Manual bed leveling using screw and paper sheet - autoleveling is just a fine tune feature

zipippino, 2017-08-28:
Do you have a video or textual guide to do that?

Sent by iPhone   
Francesco Buccoliero aka zipippino
Inforge.net Staff & Graphic Designer

luc-github, 2017-08-28:
the manual bed leveling wizard is available in the printer menu 

there are toons of youtube video for this and several post for Davinci and even more for others printers - just google it and you will get them -  this is not specific to Davinci

luc-github, 2017-10-04:
Close issue as no update

closed 2017-10-04 by luc-github