luc-github/Repetier-Firmware-4-Davinci #268: bed level keeps dropping

tecan, 2017-09-30:
im having issues with the davinci saying 620 deg then it goes to def and im unable to move anything unless done from the menu. the problem is the bed level keeps dropping when the unit comes on and then eventually bottoms out and keeps trying to go down further.

luc-github, 2017-09-30:
davinci saying 620 deg looks like  sensor issue

luc-github, 2017-10-01:
which version of Fw do you use?

tecan, 2017-10-01:
i got the sensor issue fixed, the bed is fine now but , it would also be nice to disable the "no movement when sensor not coupled issue" so that i can use laser or dremmel to do the work. it was 0.92.10m

tecan, 2017-10-01:
also when changing filament i have to reboot machine manually if the change filament error comes up and trying to load it in using pronterface. it would be nice if i pressed the home button it would let me extrude still.

luc-github, 2017-10-02:
to use Davinci as cnc you must compile fw accordingly but I never tested, it is somewhere in configuration.h look for current setting PRINTER_MODE_FFF

luc-github, 2017-11-13:
I guess this can be closed now

closed 2017-11-13 by luc-github