luc-github/Repetier-Firmware-4-Davinci #269: Filament pulled back before printing after load

TComp, 2017-10-06:
Hello,

not sure if this is a firmware issue or it's me doing something wrong. I have a Davinci Duo with the latest firmware and use Cura 2.7 as slicer. The issue happens only at the first print after loading the filament. After starting the print, the bed moves up to the right position and ther extruder moves to the starting point to print the skirt. Right at this moment, the feeding stepper quickly pulls the filament back by about 10-15 mm and then starts pushing it forward. Obviously, no plastic comes out of the nozzle and nothing gets printed. I have to stop the print, push the filament manually forward back into the nozzle and then restart printing. This does not happen the second or subsequent times or after I restart printing from scratch.but only after loading the filament. Thanks for the support.

luc-github, 2017-10-06:
Check your GCODE to see what is happening - printer only execute what you ask at this time - if the filament is back about 10-15mm it means there is a command in your model asking for this - if you do not know what command do it,  I suggest you to load filament and to send manually one by one each command of your object to find which command doing it

TComp, 2017-10-06:
Thanks for the prompt reply. I will run test you recommended and will see what happens. The reason why I posted the issue here is that if I stop the print and then restart it from the beginning, it does not happen again, even if the GCODE is always the same. Will let you know.

luc-github, 2017-10-06:
Yes that is why need to know which command is doing this time to time - it will help to see what is happening

TComp, 2017-10-25:
I have been testing my printer all this time and cannot find a pattern when and why this issue happens. I did run the GCODE one command a time and there is no command that causes this. It happens right before the start of skirt printing:

M117 Start
M300 s1000 p400 ; Beep
;LAYER_COUNT:113
;LAYER:0
G0 F2400 X52.047 Y76.227 Z0.3
--> It happens right here
;TYPE:SKIRT
G1 F900 X52.5 Y75.939 E0.02678

The problem is that it is a very rare problem; I have the impression that it happens after I pause and cancel the print, lower the bed and then restart from scratch, but have not been able to confirm it. The extruder moves to the starting point and, before starting printing, the feeder stepper spins backwards for a 3-4 seconds and pulls the filament out of the nozzle. I have been able to see this happening right in front of my eyes. Any idea what else should I test?

luc-github, 2017-10-25:
It is not related but : 
>G0 F2400 X52.047 Y76.227 Z0.3
because speed are different between XY and Z  the usage is to never mix XY and Z command, also G0 = G1 for repetier FYI

If happen after pause or cancel - check if your host does not have macro (before pause / after pause/ etc...)

it may be something with your start code also : 
here mine on S3D for Information  (temperatures are handled by S3D so not in my start code):
```
M116;wait all temperature to be reached
G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode
G28 ; Home extruder
G1 Z15.0 ;move the platform down 15mm
G92 E0 ;zero the extruded length
G1 F200 E3 ;extrude 3mm of feed stock
G92 E0 ;zero the extruded length again
M100
T0
G4 P2
;Indiana
M300 S1318 P240
M300 S0 P120
M300 S1396 P120
M300 S1567 P120
M300 S0 P120
M300 S2093 P960
M300 S0 P120
M300 S1174 P240
M300 S0 P120
M300 S1318 P120
M300 S1396 P960
M300 S0 P240
M300 S1567 P240
M300 S0 P120
M300 S880 P120
M300 S987 P120
M300 S0 P120
M300 S2793 P960
M300 S0 P240
M300 S880 P240
M300 S0 P120
M300 S987 P120
M300 S2093 P480
M300 S2349 P480
M300 S2637 P480
```

TComp, 2017-11-06:
Sorry for the late reply, but I have been testing my printer all these time to understand the pattern when this issue happens. I can confirm now that it happens always after this sequence: pause print job in Octoprint --> cancel job in Octoprint --> load filament from the printer's display (Maintenance menu) --> start the job again from the beginning. I had to stop the job a couple of times because no filament was coming out of the nozzle and, after pausing and cancelling the job, tried the load some filament to check if the nozzle was clogged. Right after loading, I tried to start the job again from the beginning and, right after the bed (Z) moves to home position and the extruder moves to the starting position, the filament feeder spins backwards for about 3-4 seconds pulling the filament out of the nozzle.

I use Octoprint 1.3.5 (latest version) on a Raspberry Pi with OctoPi 0.14 (latest version). It does not seem that it has any macro before/after pause.

The line you were pointing to was created by Cura (which I use as slicing software) and here is my standard starting code:

;FLAVOR:Repetier
;TIME:9718
;Generated with Cura_SteamEngine 2.7.0
M190 S60
M109 S190
G90 ; set absolute coordinates
G28 ; home all axes
T0 ; select extruder 0
M117 Start
M300 s1000 p400 ; Beep
;LAYER_COUNT:113
;LAYER:0
G0 F2400 X51.899 Y66.11 Z0.3
G0 X51.512 Y66.481
;TYPE:SKIRT
------------------- Here is where the problem happens ---------------------------------
G1 F900 X51.899 Y66.11 E0.02675

luc-github, 2017-11-06:
well for me something obvious is missing in start code: 
`G92 E0 ;zero the extruded length`

When you restart and ask extruder to go to  E0.02675 -  because you did not reset the length -  the extruder is reversing from the length is currently may be (+4 mm after you print and do loading ) to 0.02675

TComp, 2017-11-13:
I added the code line you proposed to the start code and the problem did not show up again, but I have been able to test it only once since I am actually out of town. Will repeat the test next week when back and will let you know. Thanks for the hint.

luc-github, 2017-11-28:
I close issue then

closed 2017-11-28 by luc-github