luc-github/Repetier-Firmware-4-Davinci #270: z height positioning bug

tecan, 2017-10-26:
when i do a manual leveling it makes my hotend about 3.8 mm too high, when i set the zminimum in firmware settings it works untill printer reboots, once the optical switch hits it clears the zheight to the mimum and shows -3.8mm but its actually not.

tecan, 2017-10-26:
turns out you have to do manual level after setting the zmin height. might have better results

luc-github, 2017-10-27:
Printer Manual leveling consider Zmin  = 0 as it should be and do not modify Zmin value in EEPROM/Memory

tecan, 2017-11-04:
i have a different hotend on it now, i need to make the zmin about 30 mm lower. for some reason the eeprom is hooped, when i delete the one from sd card it seems to use an internal one. when i try saving the settings with or without the sd card it wont save for some reason. sometimes i get lucky though. is there a clear eeprom function for the due somewhere ?

luc-github, 2017-11-04:
Due does not have EEPROM it use virtual EEPROM which is actually a file on SD card -  the file is eeprom.bin - so if you try to save setting without SD Card it will do nothing but keep settings in memory until next restart -  file is sync with EEPROM every 30 sec so if you save and immediatly restart printer settings may not be saved

A clean EEPROM is done by deleting eeprom.bin and then do a save fail-safe settings and of course wait more than 30 seconds before restarting printer

luc-github, 2017-11-13:
Any update ? can we close ?

luc-github, 2017-11-28:
close as no update

closed 2017-11-28 by luc-github