luc-github/Repetier-Firmware-4-Davinci #272: Connection failure when printer swith on after

Hawat3000, 2017-11-18:
Hello
I have a strange problem. If I power on the printer before the operating system has loaded everything works fine, but if I connect the printer after the system starts I always have a error in Repetier-Host, cant connect to the COM port. I have the same problem if I restart the printer (power off and on) when I am working in the operating system).
I have tested it in Windows 7 x64 and Windows 10 x64 (in the same computer)
Any ideas?

luc-github, 2017-11-28:
No, never got such issue sorry

Hawat3000, 2017-11-29:
Thank you for your response, and by the way for the firmware ;)
It is very strange. The com port is the same if a switch on the printer before or after. But the behavior is different.

luc-github, 2017-11-29:
did you patched the USBCore.cpp ?

Hawat3000, 2017-11-29:
If you refer to copy the file from:
X:\Repetier-Firmware-0.92-master\src\ArduinoDUE\AdditionalArduinoFiles\Arduino - 1.8.0 -Due 1.6.8\Arduino15\packages\arduino\hardware\sam\1.6.8\cores\arduino\USB
To:
C:\Users\XXX\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.8\cores\arduino\USB

Yes I have done.
I just have repeated the process two times and I have the same problem.
I know that is not a great problem (because it works if you connect the printer before) but now it is curiosity

[image]

The icon show that the printer is connected but a messagebox appears telling conection error and the printer not responds.

luc-github, 2017-11-29:
To be honest davinci USB is not the best because Due port is native port and can be peaky
Also I saw also repetier host have some com issue with windows time to time but not systematic like you

2 things you can try : 
1 - update repetier host yours is 1.06 current is 2.05 
2 - try without USBCore patch : unistall Due 1.8 from board manager then install again, do not forget to update variants.cpp (this one is mandatory),  the usbcore patch allow safer communication avoiding dead loop and printer reset if serial is disconnected when open (pc go standby, cable unpluged, etc..) this patch may disturb your com port, I am not confident about this but it worth to try ^_^

Hope it helps

Hawat3000, 2017-11-29:
Without the usbcore patch the printer still not connect but when you click on connect or disconnect the printer reboot
:-O

luc-github, 2017-11-29:
yes that is why I wrote it ^_^
it looks like something lock your com, you do not use hub ?

Hawat3000, 2017-12-04:
I have tested it with different Repetier Host, with a usb hub and  without a usb hub with the same behavior...
Can you tell me a simple "hello" to try with the Arduino IDE serial port console?

Thank you

luc-github, 2017-12-04:
it is serial communication emulation so must not use hub on this com
check in arduino terminal if you can see the printer message
also be sure repetier host has same baud rate as printer

no sure what you mean by simple "hello"
Is it that ? https://www.arduino.cc/en/Serial/Write

Hawat3000, 2017-12-04:
Sorry I have explain wrong. I mean some kind of hello command in the Arduino serial monitor to get some kind of message from the printer.

luc-github, 2017-12-05:
When printer wait for command it send "wait" message constantly

in serial monitor select cr/lf 
You can send M115 to get printer info, M105 for temperature M20 for file list , M205 for EEPROM output
check list here : https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/src/ArduinoDUE/Repetier/Repetier.ino

Hawat3000, 2017-12-11:
Same behavior, if the printer is connected when operating system starts I see the wait command in the Arduino Serial monitor. If I switch off the printer and switch on again the wait command disappear

luc-github, 2017-12-11:
after switch off printer you need to close then open again Arduino Serial monitor because connection is closed. Then you should see wait command.

Hawat3000, 2017-12-11:
I have closed and restarted the Arduino and no wait command....

luc-github, 2017-12-11:
strange - what about usb unplug cable then open arduino terminal ?

Hawat3000, 2017-12-11:
same behavior....

luc-github, 2017-12-11:
I never had such issue before but gave a try today and was able to reproduce it
I have just updated from win7  to windows 10 a week ago not sure it is linked 

Arduino terminal says port is busy!!! and repetier host cannot connect to printer!!!

So I have uninstalled due drivers and restarted computer as requested  (printer is not connected) - then I have connected printer 
Arduino terminal show nothing when connected to printer 
But now Repetier can connect 

Not sure where the issue come from but I suspect the due module or the drivers

I set this as bug and will try different modules to see if it is solved with latest or if it is a regression issue and check old ones

luc-github, 2017-12-11:
After updating to Due 1.6.11 and of course updating variant.cpp /USBCore.cpp I cannot reproduce anymore 

Can you try and feedback ?

Hawat3000, 2017-12-11:
When you says updating to Due 1.6.11 do you refer to Arduino 1.6.11 or card
version in Arduino cards management?

luc-github, 2017-12-11:
due board in board manager, my arduino is still 1.8.4

Hawat3000, 2017-12-12:
Now I have a big problem.
I have done what you told me, but now the 3d printer has bars in the screen and Windows not recognize the printer.
The usb connects and disconnects constantly so I cant upload the firmware again.

luc-github, 2017-12-12:
that means wrong location for variant.cpp and USBCore.cpp - you have to use jumper to reset

Hawat3000, 2017-12-12:
I am trying to reset via jumper (i did it two years ago), I am trying now but it doesn´t work.
Please tell me if i am making something wrong.
1) with printer power on (without usb) jumper pins JP1 during 5 seconds.
2) switch off power and switch on again
3) My black bars continues on screen and when i connect it to the pc, the usb connects and disconnects

Hawat3000, 2017-12-12:
Ok, now it is reseted

luc-github, 2017-12-12:
JP1  so you have a Davinci 1.0 then ? the one with the normal SD card not micro one
I have Duo with same board base and this one is more robust than new generation so no worry there is few chance you briked it

Hawat3000, 2017-12-12:
Some more tests:
- Arduino 1.8.0 and  Due 1.6.11 (updated from 1.6.8) in windows 7, resut: the same behavior
- Arduino 1.8.4 and  Due 1.6.11, fresh install and windows 10, resut: the same behavior

These are my Arduino Due drivers:
[image]

luc-github, 2017-12-12:
I have same drivers, Arduino 1.8.4 and Due 1.6.11 on windows 10

PC is on - printer is off and disconnected
Printer On then connect : Arduino Serial terminal and repetier host are OK ( can see wait msg and connect)

Power Off Printer still connected - Power On printer : Arduino Serial terminal and repetier host are OK ( can see wait msg and connect)

I am not sure what is the problem as now I cannot reproduce it anymore ...

Hawat3000, 2018-02-03:
Hello Luc
I just have tested it in a third computer and the problem stops. On this PC
you can switch off and switch on the 3d Printer without problem.

luc-github, 2018-02-03:
so it is confirmed it is on PC side - thank you for the feedback, I close issue then

closed 2018-02-03 by luc-github