luc-github/Repetier-Firmware-4-Davinci #273: Filament not fed with one extruder only

TComp, 2017-11-25:
I am making some modifications on my DaVinci Duo with the latest Repetier and have removed the left extruder T0 and want to print with T1 extruder only (placed at the center of the carriage).  The heater, thermistor and fan of T0 extruder are all disconnected (decoupled). I can load and unload the filament on T1 without any problem. The issue is that when I launch the print, the bed and extruder move correctly on X, Y and Z axis but the filament feeder stepper does not move at all and does not feed or retract the filament. Does this happen because the T0 is missing?

luc-github, 2017-11-25:
You may left some GCODE to heat T0 in your start code - which cause issue - remove every reference to T0 - it should work

TComp, 2017-11-25:
Thank you. The reason was a T1 command missing. I had T1 going to the right temp but then had forgotten to add another T1 below it and the printer was defaulting to the T0 extruder. Thanks for the help.

closed 2017-11-25 by TComp