luc-github/Repetier-Firmware-4-Davinci #274: Extruders showing different temperature

TComp, 2017-11-27:
I have replaced the original hotends with E3D V6 clones with new heating elements and thermistors. I am using the default code #5 for thermistors  (custom table 0) for both extruders. The two thermistors are showing an 8C difference at room temperature: room temperature approx 22C, bed temp 22.4C (original XYZ sensor), ext0 temp 24.1C, ext1 temp 16.1C.
First I thought it was the thermistor but the problem was still there after I changed ext1 thermistor. Then, I swapped the two thermistors, but the temp difference remained the same with ext1 still 8C lower than ext0. Both thermistors are NTC 3950 type with 100K resistance at 25C. Any suggestion what it might be? Should I used code 14? Still, why the difference if both are configured the same way?

luc-github, 2017-11-27:
https://github.com/luc-github/Repetier-Firmware-0.92/issues/104

TComp, 2017-11-27:
Thank you, but I still do not understand why showing different temperatures when using the same thermistor and same parameters. If there is a dicrepancy, it should be the same on both, correct? Don't you think that even with custom tables the difference will still be there?

luc-github, 2017-11-27:
because thermistors are not really 100% identical and davinci circuit may not have 100% identical resistance on T0 and T1, Davinci use cheap component which may have 5% or more tolerance -  even on my Davinci with original thermistors I have 1~2 degres difference at room temperatue. 
Unlike working temperature (190-230 , room temperature is in very sensitve zone of table:
[image]

So even small difference in measurement will have huge impact in temperatures rendering, and because this table is an experimental table, due to tolerances I explained before same temperature won't give same ADC measurement and so same Temperature display.    

If you ask me if it is FW issue I will say no -  it is hardware - Original Davinci does not display room temperature - juste heating temperature if I remember well, which make sense as they are not able to do it

TComp, 2017-11-27:
I am sure this is not a FW issue, just wanted to understand. The difference between thermistors does not have any effect here because I am connecting the same thermistor to both Ext0 and Ext1 wires and it shows a difference of 8C. The difference remains the same even if I use another thermistor. It should be the resistors on the board even if it is hard to believe there is a 33% diefference at room temp (8C over a temp of 24C).
I prefer to remove the resistors on the board; do you happen to know which are the 1K resistors on the Duo board? Also, can you please confirm that the resistors should be simply removed by leaving the terminals open instead of bridging them? Thanks.

luc-github, 2017-11-27:
I never did the operation myself so I won't be a good support - and about bridge vs open : open is correct  as pointed here https://github.com/luc-github/Repetier-Firmware-0.92/issues/237

TComp, 2017-11-27:
I know that on DaVinci 1.0 the resistor is R29 but don't know which resistors have to be removed on the Duo board. I will find them, thanks anyway.

luc-github, 2017-11-27:
Duo and 1 have same board and component position should be side by side  so once you foud R29 for T1 (T0 on davinci 1) you should easily guess the T0 components

luc-github, 2017-11-28:
I close as not FW related

closed 2017-11-28 by luc-github

TComp, 2017-11-28:
Ok, thanks. Just wanted to let you know that I found the resistor to be removed for Ext1 thermistor. The resistor is the R200 (see attached image). Removing R29 for Ext0 and R200 for Ext1 and configuring the two thermistors as type #8 (Semitec 104GT-2) in the configuration.h file should allow NTC 3950 thermistors to be used with the DaVinci board.

I sent this message directly to you since you closed the thread.

Thank you again.

From: Luc
Sent: Tuesday, November 28, 2017 2:30 PM
To: luc-github/Repetier-Firmware-0.92
Cc: TComp; Author
Subject: Re: [luc-github/Repetier-Firmware-0.92] Extruders showing differenttemperature (#274)

Closed #274.