luc-github/Repetier-Firmware-4-Davinci #276: Nozzle cleaning sequence and load/unloadposition

TComp, 2017-11-30:
This is not a FW issue but just a question: I have installed two E3D V6 clones and completely changed the configuration of the extruders' carriage. The two nozzles now are about 24 mm from each-other compared to 35 mm in the original configuration, but always centered in the carriage (12 mm from the carriage X center). Y positions are about 3-5 mm off center and Z is still the same. Right now I have a -13mm X offset for Ext0 and -37mm for Ext1 (they are dead center at X=75mm and Y=100mm).

The main problem is that when loading/unloading the filament, the nozzles are right at the edge of the dripbox and one of them exatly on top of the cleaning blade. Also, when running the nozzle cleaning sequence, both nozzle miss completely the cleaning blades. Any hint how to fix this?

luc-github, 2017-11-30:
Check Clean noozle function which use the blade position defined in configuration.h
https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/src/ArduinoDUE/Repetier/Configuration.h#L87-L90

Function is there : https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/src/ArduinoDUE/Repetier/Printer.cpp#L209

for load / unload it use xmin and bed size so you need to change position in function itself:
https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/src/ArduinoDUE/Repetier/Printer.cpp#L209

so to sum up check the code and change it according your needs :smile_cat:

TComp, 2017-11-30:
Thanks. I was a little confused because in another thread somewhere else you said that the values in the Clean Nozzle function are distances and not positions. I will start playing with different values and see how it works. Thanks again.

luc-github, 2017-12-01:
Well, yes they are distance from Xmin/Ymin actually - if you check the code it is obvious

TComp, 2017-12-01:
Thank you for the info.

TComp, 2017-12-06:
I just want to make sure I am understanding correctly: you are pointing to the same cleanNozzle function for the load/unload extruder position. Is that correct?

luc-github, 2017-12-06:
No sorry I did wrong paste and did not noticed it - i have upaded the link

luc-github, 2017-12-10:
I guess this issue can be closed as it is edge case

closed 2017-12-10 by luc-github