luc-github/Repetier-Firmware-4-Davinci #279: XYZ 1.0 AIO Extruder doesn't work only when printing

loneacoustic, 2017-12-27:
I can load and unload and extruder works fine, but when I try to print nothing comes out, makes a clicking sounds sometimes, I have checked everything regarding hardware, tried cura, simplify3D and repetier host, I will try to reflash everything again but I need help plz, also if there any way I can help get the scanner working I'll do any testing and/or if I can any translations

luc-github, 2017-12-27:
Looks like you need to do bed leveling - your bed is too close to nozzle at first layer so extruder cannot extrude properly

loneacoustic, 2017-12-27:
I have purposely left the bed very low, the nozzle is  not touching the bed, no filament coming out at all only when printing, also I have the control flow to hardware so I can connect to S3D others are stock cura and repetier, none work still

luc-github, 2017-12-27:
then if you can load / unload properly and it is not a bed issue  - the issue come from the GCODE 
your start GCODE do not wait temperature is reached may be ?

loneacoustic, 2017-12-27:
I thought the same thing, it's odd that it works perfectly fine thru S3D when using jog controls to extract some filament, I checked the temps too and they are good, I used an IR thermometer to check temps as well, I will try once more before I flash the whole thing, I just flashed the firmware a few hours ago, I also just got this printer a few hours ago lol, I'm using Rhost 2.0.5 right now and printer has 0.92.10m firmware

luc-github, 2017-12-27:
share the start code of your file do not put the lines starting by ; they are just comments and useless

loneacoustic, 2017-12-27:
For S3D and all of them i used the default Gcode, you know G28, probe homing and all that, and at the end turn off temps and disable steppers, not at home rn but what should my Gcode look like in S3D

luc-github, 2017-12-27:
instead of click send over USB use the second button save to disk then edit file in text editor

loneacoustic, 2017-12-28:
I have found my issue, the stepper motor cable is bad, when it's on home it does extrude because it's making contact, but once a print runs, moving back and forth, it loses connection and clicks (many would think it's a clog or it's too closed to the bed) but it's the cable. After replacing the cable it works great, hope this info helps anyone with this issue. And thank you for your help, firmware and continued support, will definitely donate, can't wait to have the scanner running hopefully*

luc-github, 2017-12-28:
Great if you found the root cause - yes Davinci wires / connectors are pretty fragile

For the scanner, as I do not have an AiO and the original developer of Horus seems not doing maintenance on code any more - I am not sure when a solution will be found -  but current horus code is there so nothing prevent to continue.

Happy printing and Happy New Year 🎆 🎉 🎈

closed 2017-12-28 by luc-github