luc-github/Repetier-Firmware-4-Davinci #280: Da Vinci 2.0A Duo (j37)

TrollCtrl, 2017-12-29:
Hi I recently flashed my Da Vinci Duo 2.0A with repetier firmware, step by step, as it say in manual.

`// ################## EDIT THESE SETTINGS MANUALLY ################

#define DAVINCI 3// "0" if not DAVINCI, "1" For DAVINCI 1.0, "2" For DAVINCI 2.0 with 1 FAN, "3" For DAVINCI 2.0 with 2 FAN, 4 for AiO (WITH NO SCANNER SUPPORT)
#define MODEL  1//"0" for first generation (jumper JP1 to reset ) , "1" for new generation   (jumper J37 to reset)
#define REPURPOSE_FAN_TO_COOL_EXTRUSIONS 0 //Setting this to 1 will repurpose the main Extruder cooling fan to be controlled VIA M106/M107
                                           //Warning: for DaVinci 1.0 need to add a permanent fan with power supply to cool extruder`

If I set the position 75/150mm of X axis, the extruder is shifted to the right of the center of bed.
[image]
In the firmware settings it is said that the x-axis is 199 mm long. If I set 99.5/199mm, the extruder goes to the left of the center of the bed.
[image]
What i do wrong, and how I can fix this?

luc-github, 2017-12-29:
First the X axis lenght is including the left drop box so it is normal that half of the lenght is not in middle of the bed 
Second : the bed is supposed to be 150 mm so Yes 75mm should be middle of the bed but unfortunaly I do not have a Duo A so I was not able to verify these settings, the current settings are for my Duo but seems there is some differences with Duo A.
So to fix this you need to check what is the position when home : 
current position defined is 0, 0  if these coordinate are not correctes it is normal that the position 75mm won't be correct
Adjust the X min Ymin position in EEPROM to match real position, you may also need to modify all 
others positions/distances as well 

Looking at your picture Xmin looks around -5mm

If you change in code then reflash - do not forget to load fail safe to reset EEPROM with new values

https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/src/ArduinoDUE/Repetier/Configuration.h#L1102-L1106
https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/src/ArduinoDUE/Repetier/Configuration.h#L1082-L1086

S1CAR1US, 2017-12-30:
Hi Luc, not sure you remember me but I been around since the beginning.  I had talked to you a couple years back about the z probing with an e3d nozzle...  Now I am not sure where this has gone as far as your development, but after recently tuning up my 1.0 again, I was going to test it out by finding the pin that correlates with the signal and just trying to send a ground signal to that pin... but low and behold with just some minor offsets to the z probe x and y offsets, it already works!  So the auto leveling feature goes to all three points and picks up good values, however, after reaching the last point, it displays the value, then starts to move and then gives me a z probe failed error.  I didnt want to open a new issue because I am not ssure that it is an actual issue or if it is due to something I am doing wrong or what.  One thing is certain, the bare e3d nozzle is definately tripping the sensor and it is coming out with good numbers.  I dont know why I cant find any discussion on this.  When I do a google search for it, the old conversation I started like 2 or 3 years ago on voltivo forum comes up!  Please advise

luc-github, 2017-12-30:
@S1CAR1US this is nothing related to @TrollCtrl  issue I move your question to another ticket

TrollCtrl, 2018-01-11:
1. At the beginning of printing, reaching the left end continues to move (drive belt clicks), as it was before editing Xmin.
2. The center of the printed object still shifted to the right from the center of the table.

luc-github, 2018-01-11:
this means Xmin is incorrect or new Xmin is not taken in account, how did you modified this setting ?
What comnand make reach the left bed ?

TrollCtrl, 2018-01-11:
> this means Xmin is incorrect or new Xmin is not taken in account, how did you modified this setting ?

I set Xmin from printer menu and save settings to EEPROM.
> What comnand make reach the left bed ?

Dunno, where can I find it in simplify3d?

luc-github, 2018-01-16:
if you save setting of Xmin to EEPROM it is correct - after you must do a homing to reset the extruder position according new Xmin

So if the commands
```
G28 // home
G1 Z5.0 //down bed 5mm
G1 X75.0//put middle of bed
```
do not set Extruder in middle of bed for X - then Xmin is not correct

for the command : you read the gcode file in notepad instead of sending to printer, I hope it make sense ?

luc-github, 2018-01-25:
Did you found the correct Xmin ?

TrollCtrl, 2018-02-02:
nope, now there is no free time for this

luc-github, 2018-02-03:
No problem - I close issue then - feel free to open if needed

closed 2018-02-03 by luc-github