luc-github/Repetier-Firmware-4-Davinci #281: z probing with an e3d nozzle

luc-github, 2017-12-30:
@S1CAR1US here the ticket
Hi Luc, not sure you remember me but I been around since the beginning. I had talked to you a couple years back about the z probing with an e3d nozzle... Now I am not sure where this has gone as far as your development, but after recently tuning up my 1.0 again, I was going to test it out by finding the pin that correlates with the signal and just trying to send a ground signal to that pin... but low and behold with just some minor offsets to the z probe x and y offsets, it already works! So the auto leveling feature goes to all three points and picks up good values, however, after reaching the last point, it displays the value, then starts to move and then gives me a z probe failed error. I didnt want to open a new issue because I am not ssure that it is an actual issue or if it is due to something I am doing wrong or what. One thing is certain, the bare e3d nozzle is definately tripping the sensor and it is coming out with good numbers. I dont know why I cant find any discussion on this. When I do a google search for it, the old conversation I started like 2 or 3 years ago on voltivo forum comes up! Please advise

luc-github, 2017-12-30:
Well I still did not moded my Duo with e3d yet - if contact is Ok with the probe the fail may be due to zprobe size and final calculation which if out of some boundaries may generate the failure

check current values here : 
https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/src/ArduinoDUE/Repetier/Configuration.h#L1586-L1623

S1CAR1US, 2018-01-01:
It actually worked one time so far and asked if I wanted to store the values in for auto leveling... something like that.  I tried to find a reason why it would be failing and only thing I can find is on the rep rap wiki it says something about if the machine is bumped or shakes it might cause an error?  I dont know if that is it but I recently switched back to a direct drive system so my x axis is lugging around that giant stepper again.  I have a titan extruder and pancake stepper coming in the next few days so maybe that will help.  Anyways, even with the error happening after the third measurement, it is still capable of being useful for leveling the bed in the same way which the original xyz ware does, in that it gives you the three measurements down to I believe a thousandth of a mm.... so just gotta do it a few times, take note of the measurements, and end up with a pretty dang good bed leveling.  Its the nicest I have had it since I first switched over and got rid of the xyz firmware!   

Also: I think that what your saying may be it on second thought, because i did see in the firmware that there is a max of 5mm for the probe distance.... I forget exactly what the variable is named, but I was getting measurements just barely over 5mm.  So that should be an easy fix, either move the bed in hardware or see if it will allow a higher value in firmware.  I will report back after some more testing.  

Thanks for the quick reply like always!  Sad to admit I have a MKS Base v1.4 on the way and will most likely be moving over to Marlin....  You have provided a great firmware port through the last few years and saved me so many headaches if I had to deal with XYZ locked junkware, so want to say thank you for all of your continued effort on this.  If you have any input, resources or direction on the MKS Base board w/ davinci id love to hear your thoughts.

S1CAR1US, 2018-01-01:
One more thing I forgot about that I wanted to ask your opinion on, have you ever thought about re-purposing the top lid sensor/pin for a z probe?  As far as I understand, the reason you cant add an aftermarket sensor is because there is no open pin that works for a sensor.  Wouldn't that be an easy work around and even be able to use the same sensor?

luc-github, 2018-01-01:
Yes the 5mm limit was what I had in mind - but forget where it was or may be was before the bed leveling rewrite now all code is here : https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/src/ArduinoDUE/Repetier/BedLeveling.cpp 

about mks I do not have lot of experience on it I have a mks 1.5 which is not marlin /repetier compatible but for smoothie - there is a wifi module and tft touch screen which is very responsive per my test but never fully used the set yet

About Marlin currently I prefer Marlinkimbra flavor https://github.com/MKFirmware/MK4duo which is very nice and have more feature than marlin itself
But in the future soon there will have a marlin for 32bit MCU (the devt version 2.x) which should even support ESP32 which will bring great features ^_^  https://github.com/MarlinFirmware/Marlin/pull/7926/commits

about repurpose you can use the filament EEPROM reader pin for this there are 3 wires VCC GND and data so it should fit the need for a probe

S1CAR1US, 2018-01-02:
It's confusing because I get conflicting information all over the place which I am sure is attributed to the boards being made by cheap cloning vendors who label them wrong it misrepresent their capabilities.  The one I ordered says it has an atmega 2560 chip... Isn't that not capable of running smoothie?  Is the mks base board same as smoothie? Is there a difference between mks base and mks sbase?

luc-github, 2018-01-02:
yes they do many boards, mks-base are mega/ramps based, mks-sbase is smoothie based
check here https://github.com/makerbase-mks

luc-github, 2018-01-25:
I guess that now you moved to another board,  topic can be closed

closed 2018-01-25 by luc-github