luc-github/Repetier-Firmware-4-Davinci #282: ARduino 1.8.0 does not have Native USB Port option?

pilza, 2018-01-16:
Hi, I am trying to flash my DV 2.0A Duo with repetier 92. I got all files downloaded without a hitch etc. However, arduino 1.5.8 is the only IDE that has Native USB Port as an options, whilst  higher versions (I tried 1.6.8., 1.6.9, 1.8.0, 1.8.4) do not have the Native USB Port in the Tools -> Board menu. Obviously when I try 1.5.8 and compile the repetier code I get compile errors as expected. 
Q1: Which Arduino works with the latest repetier and has the Native USB Port option?

Q2: If only Arduino 1.5.8 has the Native USB Port function, then where can I find the repetier code that works with this IDE?

Hope you guys can help as in the meantime I am looking at a useless 3D printer.  Thanks.

luc-github, 2018-01-16:
did you checked the FAQ / wiki  as sugested in Readme ?

https://github.com/luc-github/Repetier-Firmware-0.92/issues/166

pilza, 2018-01-16:
Ugh…. I now see it as well. I feel like a monkey. Thanks!

 

#166 <https://github.com/luc-github/Repetier-Firmware-0.92/issues/166>

closed 2018-01-16 by luc-github