luc-github/Repetier-Firmware-4-Davinci #283: DV 2.0A Duo repetier flash problem

pilza, 2018-01-16:
Hi,
I have a da Vinci 2.0A Duo (Jumper37) which shows two black bars upon switch on. 
- DV switched off I shortened JP37, switched on DV, wait and switched off in order to remove any FW.
- I download Arduino IDE 1.8.0 and via boardmanager upload 1.6.8 module which allows me to select Native USB Port
- I downloaded Repetier 0.92, extract files and copy over variants.cpp and UBSCore.cpp from repetier and save into exact same folder destination in Arduino.
- switch on DV and device manager shows Port COM6 connected and called Bossa.
- I start Arduino, open repetier file from src path.
- I change configuration-h to 3 and 1 to reflect that its is a Duo two fans, and later version (JP37).
- I compile sketch and it shows all good.
- I upload sketch, Arduino shows upload progress and verification progress 100%, tells me done in 47 seconds, set boo flash true, CPU reset.
- DV made an ever so slight noise (as if one of the stepper motors had a nanosecond power supply), but nothing else happens, printer does not move and black bars remain.
- computer win10 now tells me that the last USB device not recognised, i.e. after upload finished it struggles to connect to printer.
- I did this twice, i.e. short JP37, get bossa , upload , etc.

Since the printer does get to the connected Bossa state I was certain that it will be flashable, but now not sure what is going on. The printer got into this mess in the first place when it was doing the XYZ firmware update and never finished it somehow, hence the comma.

Any ideas if I did something wrong in the process or if the USB driver is somewhat bad pre or post upload? This printer is doing my head in.

luc-github, 2018-01-16:
Did you read the FAQ about black bar : root cause and solution ?

pilza, 2018-01-16:
I did look at the FAQ called “Balck bars on LCD after flashing” and that is when I made sure I copied over the variants and USBCore files again to the correct verbatim folder destinations. I then flashed again and flash went 100% fine, but printer still has black bars and post flash does not show up in device manager as it does not recognise the USB.

luc-github, 2018-01-16:
Flash is always fine - but FW is incorrect 
what is the path of the files you updated on your PC ?

pilza, 2018-01-16:
Oh btw the variants size shows as 28573 bytes as per properties

For the USBCore it is 24086 bytes

 

In case that helps to identify versions.

luc-github, 2018-01-16:
what is important is the path 
what is the path of the files you updated on your PC ?

pilza, 2018-01-16:
C:/programfiles(x86)/Arduino/hardware/sam/1.6.8/variants/Arduino_due_x/variants.cpp

 

C:/programfiles(x86)/Arduino/hardware/sam/1.6.8/cores/Arduino/USB/USBCore.cpp

luc-github, 2018-01-16:
so location is wrong they are not in program files read the FAQ : https://github.com/luc-github/Repetier-Firmware-0.92/issues/114

pilza, 2018-01-16:
C:\Users\Gloria\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.8\variants\arduino_due_x\variants.cpp

 

Is this correct?  There is already a …\sam\1.6.11 in there. Leave that?

 

Basis the above I get compile errors:

 

 

Board Arduino_due_x (platform sam, package Arduino) is unknown

luc-github, 2018-01-16:
if you have \sam\1.6.11 instead of \sam\1.6.8 it means you are not using 1.6.8 but 1.6.11 so it won't work 
go to board manager and remove 1.6.11 - install 1.6.8, 
close arduino ide
go to update the variant.cpp and usbcore.cpp
reopen arduino ide load fw and flash

pilza, 2018-01-16:
No what I am saying is I got both i.e. \sam\1.6.8    and 1.6.11  folders.

 

In Arduino I only have board 1.6.8 installed, no 1.6.11 at all.

 

So I wonder should I just remove folder 1.6.11 in the users…..\sam path?

luc-github, 2018-01-16:
you cannot have 2 folders, if there are 1.6.11 and 1.6.8 it is wrong : uninstall / update in IDE did not do worked properly or you did manual changes - so delete both folders- even manually to redo from scratch using only 1.6.8

pilza, 2018-01-16:
- removed all traces of arduino and reinstalled fresh 1.8.0 with due 1.6.8.
- copied USBCore.ccp from repetier src path to: C:\Users\Gloria\AppData\local\Arduino15\packages\arduino\hardware\sam\1.6.8\cores\arduino\USB\

- copied variants.cpp from repetier src path to: C:\Users\Gloria\AppData\local\Arduino15\packages\arduino\hardware\sam\1.6.8\variants\arduino_due_x\

connected to COM7 (per device manager showing bossa) board set to native USB, compiled ok, uploaded ok with cpu reset, but then again printer post upload does not connect via USB and black bars still in display. 

Do I have the paths correct above?   thanks for your patience btw.

luc-github, 2018-01-16:
it is not copied the action, it is replace existing files  - you should replace existing ones 
is it the case ? if not it means the path is wrong

pilza, 2018-01-16:
I replaced files, it shows new date and time stamp.   path is as was given by install, I did not change it

pilza, 2018-01-16:
do u want me to erase anything after 1.6.8 and replace with what is in repetier after 1.6.8? i.e. replace whole tail?

luc-github, 2018-01-16:
yes may be redo and copy manually each of 2 files to be sure you replace the original ones in correct path 
check file existence before and and after update - this is the only advice I can give you

pilza, 2018-01-16:
ok will try, thanks again for your patience

pilza, 2018-01-17:
I tried again a few times and even managed to load a different sketch which changed the Bossa to Arduino DUE so now I know the flashing definitely works, just a matter to flash the correct bit. I then reinstalled Arduino and when I set the 1.6.8 module I replaced repetier files as always and then I made sure that the printer was in arduino Due mode, THEN chnaged baud to 1200 to empty flash and then reflashed repetier and holla it worked. After that I did the fail safe load. and now printer boots up properly...... phew. Is this is or anything else to load via gcodes or so before i start playing with this printer.  Thanks for your patience btw.

luc-github, 2018-01-17:
now you are good to go - you may set the proper dimension from your host / slicer 

Also as you noticed USB communication can be difficult - I would suggest you to buy an SD extender and even a wifi SD card to print from SD instead of USB

Originaly Davinci print from SD - not from host - I suggest you to do same 
I personnaly use a wifi SDCard and never got issue

luc-github, 2018-01-17:
Happy printing

closed 2018-01-17 by luc-github

pilza, 2018-01-17:
Will do thanks.

pilza, 2018-01-17:
Last question…. The SD extender goes into where?  The SD slot on the PCB? If so do I then copy all of the data from the microSD onto the Wifi SD card so that the SD data is intact and the card becomes a wifi rec/trans?

luc-github, 2018-01-17:
yes

akprint, 2018-01-27:
Pilza I have the same issue could you help me out

pilza, 2018-01-27:
1) watch https://www.youtube.com/watch?v=rjuCvlnpB7M first so that you get an overview what needs to be done. https://www.youtube.com/watch?v=7vKGp77EF98 is also a good overview.
2) then make sure you use https://github.com/luc-github/Repetier-Firmware-0.92 information on the latest Arduino to use as many guides still refer to the old 1.5.8, I used 1.8.0 and make sure you then upload the 1.6.8 otherwise you will not see the native USB option.
3) Make sure you reset the firmware by shorting J37 or J1 depending on what model you have and whilst short switch on, wait 10 seconds, switch off, stil keep short and switch on again wait 10 seconds then switch off. basically do it twice in a row to be sure. remove jumpershort and switch on and you have the two black bars.
4) make sure you use a good quality USB cable.
5) follow Luc's instructions to the bone on the github section. For some the firmware upload will work.
6) if it did not work (thats what happened to me) then instead of repetier I uploaded a different sketch as a tester look at https://forum.voltivo.com/showthread.php?tid=7975 and upload the little proiject. It worked for me and by doing so the Bossa (in device manager) changed to Arduino DUE so it meant something went into the chip.
7) I then tried to reload the repetier as before but befor ethat changed the baud rate to 1200 and later back to what it was (cant remember number) the low baud rate emptied the flash but somehow I felt the connection was better and the repetier uploaded fine. I continued from there as per Luc instructions and all is fine now.

I am not saying this method works for all and might nt even make sense, but the convoluted way of emptying the flash and reloading has worked for me. I have a DUO 2.0A (Jp37) so it worked for that model.

Also have a look at the long list of FAQ from Luc link as it also helped me figure things out a bit.
Good luck

luc-github, 2018-01-27:
@pilza thank you for sharing - I will add a basic test sketch in FAQ (may be based on bgm one) but changing pin for J37 as for matchi LED pins hope this will help
```

luc-github, 2018-01-27:
//int ledPin = 85; // JP1
int ledPin = 17; // J37
void setup() {
  SerialUSB.begin(115200);
  pinMode(ledPin, OUTPUT);
}

void loop() {
  int val = HIGH;
  while(true) {
    SerialUSB.write("LED ");
    SerialUSB.write(val ? "ON" : "OFF");
    SerialUSB.write('\n');
    digitalWrite(ledPin, val);
    val = !val;
    delay(1000);
  }
```

monkeyman40, 2019-06-09:
I am also having this problem with the latest version of the firmware you have posted when downloaded on June 2, 2019. I looked through every single file presented in your file to download, but did so here on your hub to ensure what I received was a match. In neither case does "variant.cpp" file exist.  So it seems the only stated fix for my black bars that show no matter which board I select as my DV 2.0 Duo having.  The Arduino IDE claims that all is flashing without issue no matter which board I select my printer as having but nothing but black bars follow.  If the issue is the SerialUSB code you present above, where is that code placed in the many files the firmware update includes?

Thank you very much for taking the time to read my post and feel free to ask any questions you need.

Respectfully,
David

luc-github, 2019-06-09:
you means this file ? https://github.com/luc-github/Repetier-Firmware-4-Davinci/tree/master/src/ArduinoDUE/AdditionalArduinoFiles/Arduino%20-%201.8.0%20-Due%201.6.8/Arduino15/packages/arduino/hardware/sam/1.6.8/variants/arduino_due_x

monkeyman40, 2019-06-09:
No. My DV 2.0 displayed as a Due after I had Arduino IDE update the board drivers within itself and the printer immediatly switch from Bossa to DUE after that and no updates show as available when I have Windows 10 and 7 check for updates. In addition I do not see a "variant.cpp" file there nor in my download.

luc-github, 2019-06-09:
Sorry what is the issue ? you cannot find variant.cpp in my repo ? or in due module from arduino ? 
what do you mean by "no updates shows" ?

monkeyman40, 2019-06-09:
I mean that I had Arduino IDE update the board files it uses and there was an update for multiple DUE files a of which I had it install. By "...no update show" I mean that Windows finds no updates available for it's own driver for the DUE.

luc-github, 2019-06-09:
Ok so now you must update variant.cpp and USBCore.cpp as mentionned in readme and if you do not find them you have the FAQ and the pined issue on top of issue telling you where they are.
[image]

but you still do not find them ?

monkeyman40, 2019-06-09:
Ahh! It's a portion of the usbcore.cpp file! No wonder I couldn't find it. I was looking for what was said to be the issue ("variant.cpp") but in fact it's the "usbcore.cpp" file (writer wipes forehead of stress sweat and is going to replace which it currently is to what you posted above). I'll change or edit my posts after doing so if it fixes the issue.
By the way, how do I find out which board my DV 2.0 w/ JP37 and dual Extruder fans unit has? My board says (X's replace data I can't see at the moment) "xxx xxx V2" so I suspected I was supposed to select the V2 board option.

luc-github, 2019-06-09:
black bars = variant.cpp
USBCore.cpp is to avoid a reset of printer when you remove usb cable
they are 2 differents files

Davinci 2.0a has 2 fans only some 2.0 have 1 Fan - they where the first batch I guess

monkeyman40, 2019-06-09:
Black bars again...
I copied the new USB code you presented above and replaced everything with it but the same result occurred. By the way my board says "F1 x F2  x Engine board V2".

N129BZ, 2019-06-09:
If you're using Windows, you'll find both both files if you search under
[drive]:\users\[your user name]\AppData\

Include subfolders in your search

N129BZ, 2019-06-09:
Actually they're under AppData\Arduino\

Do a search for each file name, replace those with the ones from the
firmware zip.

monkeyman40, 2019-06-09:
I found them where I had saved them instead which is how I altered the 2 variant.cpp files to his text above a few posts. Didn't fix the problem.

monkeyman40, 2019-06-14:
I replaced the hardware folder in the Arduino program folder (c://Program Files x86/Arduino) then reset the printer again then tried to install the firmware yet again. Things were made far worse. Now the left blocks on the upper and lower bars in the printer display no longer show and yet again the flash didn't work. The only things I changed was settings in Configuration.h to #define DAVINCI 3, #define MODEL 1 and placed the exact USB text luc_github posted above replacing all other text the variant.cpp file even though it  doesn't include the standard Arduino /* start nor any standard end of file notation apart from }. I copied the entire contents of the \src\ArduinoDUE\AdditionalArduinoFiles\Arduino - 1.8.0 -Due 1.6.8\Arduino15\packages\arduino\hardware folder into the Arduino program folder.
The new problems occured.
What now?

monkeyman40, 2019-06-14:
I tried to lookup WINSTAR screens in the Arduino Libraries and suddenly the cause of the problem was displayed by Arduino IDE... I copied the hardware folder over exactly as N129BZ said and that is the cause. Here was the result shown in the terminal at the bottom of the Arduino IDE:
"Could not find boards.txt in C:\Program Files (x86)\Arduino\hardware\sam\1.6.8. Is it pre-1.5?
WARNING: Error loading hardware folder C:\Program Files (x86)\Arduino\hardware\sam
  No valid hardware definitions found in folder sam."
I only attempted to do this firmware update in order to provide the Da Vinci 2.0 Duo to a local grade school. They asked that the firmware be updated first so that any filament they wanted to use could be and they wouldn't flash the cartridges, or else they did not want it. I make nothing as a result of this donation except finally having an open space to put my brand new Ender 5 that has never even been powered up due to having no place to put it and my return period expires for it in just 6 more days.  I'd very much like to donate it but unless it can have it's firmware updated my only option will be to dismantle it and hope to extract any silver or gold from the board, in effect destroying it.

How is this issue resolved?