luc-github/Repetier-Firmware-4-Davinci #284: Winstar for version 1.0

S1CAR1US, 2018-01-18:
Hey there @Luc,

So I have just right now, literally just this minute "finsihed" with my controller board swap.  It took way longer than I expected and I have a few details to sort out yet but there is some definite improvements.  Most noticeably is the sound that the motors make.  I have read a lot of people explaining this and about how much of a difference it makes, but I wasnt getting it til i experienced it for my self.  You can literally get it to run close to silent!  Of course when outputting this low of a current, there are trade offs you have to deal with, such as skipping steps.  Either way though I am fairly happy with it considering the board that I got is not the one that I originally wanted and I will likely switch it out again once I can afford a 32 bit board like the duet or smoothie.  I also decided to stick with repetier being that as of now there really isnt much in the way of features that I am gaining by switching over to Marlin.  Also this way I am able to more accurately use all of the parameters which you have defined over the years in your port.  This also leads me to the issue that I am working on next.  

I would greatly appreciate if you could give me some help with this part...  I am wondering how much it would take to be able to integrate the WinStar display and buttons into my firmware version.  I know that you figured it out obviously with your version, but how much would it take to get it ported over to Repetier 1.0?
Secondary to this and not nearly as important, would be the LED lights?  I cant find much about what to do to get these working. Any idea of what the pin I would use is?  How to integrate in firmware?

I am putting any files I think might be relevant to all of this in the folder which I have shared [## **here**](https://drive.google.com/open?id=1KSmZh2hp2JEJn9SGdCLReYP1qrxaPOcy).  As I keep researching I will keep adding anything I find... Thanks so much for your time.

-@S1CAR1US

This should be the only information that you would need which is relevant... let me know if anything else...
-Controller Board: MKS Base v1.4 (basically a ramps with integrated stepper drivers)
-Firmware: Repetier 1.0 ([here](https://drive.google.com/open?id=1KSmZh2hp2JEJn9SGdCLReYP1qrxaPOcy))
-Config.h: [here](https://drive.google.com/open?id=1KSmZh2hp2JEJn9SGdCLReYP1qrxaPOcy)
Punch List (ToDo): 
-WinStar Display, 
-LEDs for environment lighting and status indicators

luc-github, 2018-01-19:
Yes I know MKS 1.4 
But how you plan to connect the screen / keypad ? - it is a slim ribon connector

If you integrate winstar I guess crystal lib is the one that should be used but need to set the RW pin to low

the key pad key can be defined in https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/src/ArduinoDUE/Repetier/uiconfig.h

Led is 3 pins led and use 12v if I remember well - voltivo forum is the source for this - but you can also check the soliforum forum for the porting

Not sure I have answered your questions -  let me know if you need more

S1CAR1US, 2018-01-20:
I have been able to solder extensions off of the ribbon cable itself.  I am guessing that it's necessary to break out from there rather than from before that in the circuit?  so basically your saying it's going to be more work than just adding the libraries and modifying the firmware?  is there any other way I can contact you about this or would you rather I just do it here?

luc-github, 2018-01-20:
on the ribbon you have these traces (but not in this order) :
Vcc
GND
back light screen
up key
down key
left key
right key
Ok Key
Home Key
Screen Enable
Screen RS
Screen RW
Screen D0
Screen D1
Screen D2
Screen D3
Screen D4
Screen D5
Screen D6
Screen D7

May be more GND VCC traces -  I think I covered all

But I have no idea of their position on ribon  -  that is the additional work that you should do to be able to connect it to your MKS, once you know their position it is peace of cake as just update uiconfig.h setting the pins

luc-github, 2018-01-25:
did you had success finding the pins ? I think what you do will help many people

luc-github, 2018-02-03:
I close issue - feel free to open again if needed

closed 2018-02-03 by luc-github