luc-github/Repetier-Firmware-4-Davinci #285: Max Extruder Temp 285°C

The-Experimentalist, 2018-01-25:
Hi 

Been a long time. I have an E3D hot end in my DaVinci AiO 1.0

The E3D-v6 as sold is limited in temperature by the thermistor to 295*C but can go higher (tested up to 420*C) with a thermocouple upgrade.

I am using Repetier-Host V2.0.1 and I can setup the printer Max extruder temp to 295°C okay. I also note in the Repetier Log that it appears to send this to the printer:

00:01:43.054 : N1800 M104 T0 S295*22

However the printer will only accept a maximum of 285°C

I can't find anything in the EEPROM settings limiting it.

Any ideas?

luc-github, 2018-01-25:
this setting is not in eeprom but is in configuration.h you must modify it and reflash

The-Experimentalist, 2018-01-25:
Just looked in my Git history, looks like I changed it before. Oh to have a memory. Thanks for your help as always.

luc-github, 2018-01-25:
Git is good to track change unlike our brain 😉 

@The-Experimentalist May I know what filament do need such high temperature ?

luc-github, 2018-02-03:
if solved I close issue then

closed 2018-02-03 by luc-github

The-Experimentalist, 2018-02-05:
Sorry been away. Only just saw your question. Trying to print PMMA from Rigid Ink with the E3D hot end. Sorted the printer but still not had time to try again. Was getting close at 285°C but not quite hot enough. Will need a thermocouple if I have to go any hotter than 300°C, right at the limit of the thermistor.

The-Experimentalist, 2018-02-05:
https://rigid.ink/blogs/news/why-you-should-try-pmma-filament-and-how-to-print-it