luc-github/Repetier-Firmware-4-Davinci #286: Sync with Repetier V 1.X

luc-github, 2018-01-27:
Since January 2018 repetier 1.0 is officially released
I think it is time to redo a full resync and then rename repository to repetier for Davinci

juri74, 2018-01-28:
That's absolutely great! :D

azoelharman, 2018-02-04:
Hope to see the new version of repetier firmware for davinci, to being more great.
Thankyou budy.

gdampf, 2018-02-16:
That would be really a step forward!

luc-github, 2018-03-01:
the branch https://github.com/luc-github/Repetier-Firmware-4-Davinci/tree/devt
Contain the original repetier 1.0.1 working on Davinci in alpha state
no customization at all : 
* no Specific Davinci command (M100, M50)
* no Custom Menu (Manual Leveling, Autolevel output, Clean Noozle, Filament load/unload, Move Advanced/Easy mode, etc...)
* no emergency stop from key pad
* no LCD output modification - original is for 20X4 but Davinci is 16X4 so currently some info is not well displayed
* no fancy sounds
* no extra settings in EEPROM instead of hard coded
* no full access to EEPROM setting from printer menu
* no specific code for WiFi connection
* no sensors support (Filament, door)
* AiO turn table support
* No specific lcd support - use 4bits lcd output
Basicaly it is full repetier (not fully tested) but should be able to print, I will continue integration later

luc-github, 2018-03-01:
Feel free to test and feedback

rhinorap, 2018-04-16:
I would love to help you on this - but your branch is perfect - I don't want to risk loosing what I've done with tweaking and minor adjustments.
If and when I get a second Duo 2.0 i'll gladly test and report back.

luc-github, 2018-04-16:
Which branch ? Master ? Or 1.X?

z1cb4, 2018-10-15:
Hi, is there any progress ? Is it possible to help you in some way?

luc-github, 2018-10-15:
No as no one feed back - I guessed no one was interrested to so I use my time on others projects
Now 1.02 is available and 2.0 is in devt so this 1.x may need update

gdampf, 2018-10-16:
Oh, no! A lot of us are very interested in this project. Please proceed!

luc-github, 2018-10-16:
@gdampf what do expect in 1.X vs current one ?

luc-github, 2018-12-09:
I do not think I will spend time on this as said no reason to spend time on update that may not affect Davinci Printer

I will review this when 2.0 is stable and move to 2.0

closed 2018-12-09 by luc-github

MasterCATZ, 2020-07-23:
so devt will be your latest ?