luc-github/Repetier-Firmware-4-Davinci #287: Jr Port Assistance

OrangeFlare, 2018-02-09:
I can try and assist anyone with information from my printer as long as nothing get's bricked to try and get a Jr. Port going if anyone wants

DaVinci Jr 1.0 Pro
It's running the ATMEL ATSAM4E8E

luc-github, 2018-02-09:
Here current status:
Junior Pinout is almost known now (not idea what the Pro and Jr Pro use in addition of these pins) : 
Based on people and I shared : 
http://www.soliforum.com/topic/13769/davinci-jr-repetier-port

```
LCD (J6 LCM):
Con_Pin | MCU_Pin | MCU_SYM
1          | +5V        | N/A
2          | GND        | N/A
3          | 111         | PC18
4          | 82           | PC8
5          | U1_Pin1   | N/A
6          | 11           | PC0
7          | 38           | PC1
8          | 39           | PC2
9          | 40           | PC3
10        | 41           | PC4
11        | 58           | PC5
12        | 54           | PC6
13        | 48           | PC7
14        | 90           | PC10
15        | 34           | VDDCORE
16        | 32           | PA21/PGMD9
17        | 31           | PB3
18        | 28           | PE5
19        | 27           | PE4
20        | 25           | PA17/PGMD5
21        | GND        | N/A
22        | +5v         | N/A

SD1:
Con_Pin | MCU_Pin | MCU_SYM
1           | N/C        | N/A
2           | GND       | N/A
3           | 118        | PA31 / MCDA1
4           | GND       | N/A
5           | 116        | PA30 / MCDA0
6           | GND       | N/A
7           | 129        | PA29 / MCCK
8           | +3.3V     | N/A
9           | +3.3V     | N/A
10         | 59          | PA25 / PGMD13 / CTS1
11         | GND       | N/A
12         | 112        | PA28 / MCCDA
13         | GND       | N/A
14         | 70          | PA27 / PGMD15 / MCDA3
15         | GND       | N/A
16         | 62          | PA28 / PGMD14 / MCDA2

here the pinout Allen anb I found and validated :
Pin     Type    MCU    Pin Descriptor    PIO type    Function                                
106    MASK         PIO_PA5C_URXD1|PIO_PA6C_UTXD1    N/A                                    
105    MASK         PIO_PA9A_URXD0|PIO_PA10A_UTXD0    N/A                                    
104    MASK         PIO_PA3A_TWD0|PIO_PA4A_TWCK0    N/A                                    
103    MASK         PIO_PA28C_MCCDA | PIO_PA30C_MCDA0 | PIO_PA31C_MCDA1 | PIO_PA26C_MCDA2 | PIO_PA27C_MCDA3    N/A                                    
102    Pin    129     PIO_PA29C_MCCK     PIO_PERIPH_C                                    
101    Pin    28     PIO_PE5     PIO_OUTPUT_0    Up Key                                
100    Pin    27     PIO_PE4     PIO_OUTPUT_0    Right Key                        MCU    Descriptor    
99    Pin    10     PIO_PE3     PIO_INPUT                            62    PA26    
98    Pin    7     PIO_PE2     PIO_INPUT                            70    PA27    
97    Pin    6     PIO_PE1     PIO_INPUT                            79    PB6    
96    Pin    4     PIO_PE0     PIO_INPUT                            89    PB7    
95    Pin    2     PIO_PD31     PIO_INPUT                            87    PB12    Erase 
94    Pin    34     PIO_PD30     PIO_OUTPUT_0    Home Key                        105    PB4    
93    Pin    108     PIO_PD29     PIO_INPUT                            109    PB5    
92    Pin    71     PIO_PD28     PIO_OUTPUT_0                            112    PA28    
91    Pin    47     PIO_PD27     PIO_OUTPUT_0                            116    PA30    
90    Pin    53     PIO_PD26     PIO_OUTPUT_0                            118    PA31    
89    Pin    52     PIO_PD25     PIO_OUTPUT_0                            136    PB10    
[... 101 lines cut ...]
enable pin 3 sam 128
cw/ccw pin 44 sam 126
E1 motor
enable pin 2 sam 76
enable pin 3 sam 78
cw/ccw pin 44 sam 74
````
tp89 is the jumper to wipe the xyz FW THERE IS NO WAY TO REVERT TO ORIGINAL XYZ FW
[image]
So currently the task list is : 
* be able to reflash the mcu in charge of temperature and heat (LPC1115) and make it communicate with main mcu (ATSAM4E8E) using header for direct flash or main mcu  (ATSAM4E8E) as programmer
* identify pins lcd functions ,  which one is:  Enable, RW, RS, D0, D1, D2, D3, D4, D5, D6, D7 to have a lcd working
* As only duet FW has ATSAM4E8E support today but no screen support : code the lcd support
* Integrate the LPC1115 heat management in Duet FW because this kind of management does not exist yet
* Implement the pinout to test all previous point in real situation

So we are not yet on flashing easily and this list is the simple one if no issue is meet 🤞 

Nothing is impossible as XYZ already do it,  but need coding and test time on a printer that cannot be used one FW is wipped until full FW is ported. 

That said on my side the JR printer is in spare parts and wiped because testing how to do porting, the board is on my desk with some test code I used to find some pins, I even soldered some headers on board to test how to flash LPC1115 (point 1)

But still need lot of time and I have currently none available because really busy with others projects.

So any assistance is welcome - even sharing Pro pinout (if you know how to do it)  - but do not expect a working FW soon

OrangeFlare, 2018-02-09:
I've also discovered this
the Jr Pins that may seem N/A are actually used by the Jr Pro

luc-github, 2018-06-02:
I close issue as if fw is developped it won't be repetier but Reprap for Duet3D which support SAM4E MCU

but there will need a lot of modification to match Jr dual MCU board

closed 2018-06-02 by luc-github