luc-github/Repetier-Firmware-4-Davinci #290: WIN 7 Ultimate 64 says unknown Device until Flashing

Tibercad, 2018-03-14:
Hello,
i have the same Problem like a other User here. Sorry my english is not so good. I will try to tell what i have made before.
At the first Time i had made an Error. I have download and installed the Arduino IDE 1.8.5 but need IDE 1.5.8.  Then i downloaded the ZIP File here. Then i install the Board like here:
((https://github.com/luc-github/Repetier-Firmware-4-Davinci/issues/166))
Then i Plug the USB to the PC. The Printer has in this time the two black rows from the Firware reset.
The Printer appears at COM3. Now i am loading the ino File and make the two changes for the Printer (3 for Da Vinci Duo 2.0A, an the 1 for Jumper J37).
Then i checked the code and compiled it. The Verify ends with no errors should it be. But the two black rows are standing in the Display of the Printer. I have tried again with the same result. Then i switch off the printer an switch on.
Now the PC dont know the Hardware. The Device Manager is in a loop that prints the Hardware is not known. When i try to install the driver by hand i dont know what i shall do.
Is my Printer kompletely damaged by the Falshing or is it able to repair it?

Greetings
Tibercad

luc-github, 2018-03-15:
I suggest you to read the FAQ https://github.com/luc-github/Repetier-Firmware-4-Davinci/issues?utf8=%E2%9C%93&q=is%3Aclosed+label%3AFAQ+ for Black bars

you need to be sure you modified properly variants.cpp - check the FAQ to find the right location 

All you described is wrong variants.cpp, and you need to use jumper before flashing to recover the board - no worry it is not bricked - just wrong flash files - the jumper J37 will clean this to allow to flash again

Tibercad, 2018-03-15:
Hello,

this is a Little bit good News that my Printer is not Dead.
Which Arduino IDE do you prefer today? Some write 1.5.0, 1.5.8 and other write 1.6.(9).
I have downloaded the zip from your Side here.

Thanks for you replay here!!!!!!!

luc-github, 2018-03-15:
You can use the latest ide  but due module must be 1.6.8 because the variants.cpp is for this version
I really suggest to stick to the readme procss as most of video are out dated and check FAQ for specific details

Tibercad, 2018-03-15:
Ok, i will install in the evening the newest Arduino IDE with the Due Modul 1.6.8. And then i read the FAQ.

Thank you for your help!

Tibercad, 2018-03-15:
Sorry, but one question.
Which release auf your Softwarepackages is for me (Da Vinci Duo 2.0A)?
I see many Releases on https://github.com/luc-github/Repetier-Firmware/releases

luc-github, 2018-03-15:
none of them :  https://github.com/luc-github/Repetier-Firmware/releases is old repo and only for first generation - not for A
use current git (https://github.com/luc-github/Repetier-Firmware-4-Davinci) for all generations

luc-github, 2018-03-15:
also be aware  I have just updated travis which was broken since repository was renamed
latest release generate binaries as well which can be useful if you are not able to compile properly 
https://github.com/luc-github/Repetier-Firmware-4-Davinci/releases

Tibercad, 2018-03-15:
Hello,
first i have installed Version 1.8.5 but the structure is a little bitdifferent how it seems. Then i habe deinstalled the 1.8.5 and installed the 1.8.0 like in your Readme. The DUE Files (1.6.8) are already installed (!?!?!) when i look in the "Boardlist" and the Arduino DUE is listet in two variants how should it be.
Now you write (4. Update variants.cpp/USBCore.cpp arduino files). First i searched by the Explorer in the Repetier Files and found them. The i searched in my Installation (1.8.0) and here i found only the USBCore.cpp. The variants.cpp is not found in C:\Programme(x86)\Arduino\.....
What is here wrong? Have i done a mistake? What shall i do with the USBCore.h? This File i in the Repetierfiles besindes the USBCore.cpp.
Many Questions - Sorry.

luc-github, 2018-03-15:
yes you did not read the FAQ properly and carefully : https://github.com/luc-github/Repetier-Firmware-4-Davinci/issues/114

it is wrote in big chars https://github.com/luc-github/Repetier-Firmware-4-Davinci/issues/114#issuecomment-275509595 but ...

Tibercad, 2018-03-19:
Hello,
i have flashed the Printer yesterday again - and it works!

Now i am reading the Tutorials for configuring Repetier Host for the first Start.
I have many Questions which i must search in the Web.

Thank you very much for your help!

closed 2018-03-19 by Tibercad

luc-github, 2018-03-19:
thank you - happy printing 😄