luc-github/Repetier-Firmware-4-Davinci #291: Z probe failed?

rhinorap, 2018-03-16:
First off, I know it's 2017 - but i've just got my mitts on a 2.0A duo solely based on the fact I cold throw Repetier on it :-D
Secondly - THANK YOU to Luc for his amazing efforts in allowing us all to pull the crap that is the XYZ firmware off an otherwise good base machine :-D

Firmware is on, no issue - but i'm getting a "z probe fail" on the auto bed leveling - is this because it's out of range, or is something awry in the config ? Any help greatly appreciated ;-)

rhinorap, 2018-03-16:
I think I may of found my own answer - I performed the "z min evaluation" and stored it to the EEprom - is this the correct procedure ?

luc-github, 2018-03-16:
2017 ? no we are in 2018 😄 
first you must do manual bed leveling using screws then do the autoleveling - if distance is too big autoleveling failed
be sure corners plates and probe are clean - if no contact it will failed
check if probe do proper contact with corner plate when doing probe - I do not have 2.0A and it was reported to me that corner positions may need adjustment on some machines - positions are in EEPROM so if needed they can be modified without reflashing

rhinorap, 2018-03-16:
LOLZ - I can't believe i wrote 2017 !! *facepalm* 

Plates are nice and clean ( used the awful leveling system previously in the XYZ firmware)
The probe positions are a little out, but *just* hit the plate on initial tests - I'll adjust them later tonight as advised.

Then I'll run a manual level as you say, and then an auto.

Top job on the work you've put in - everything works so smoothly, and the steppers are super quiet compared to previously.

luc-github, 2018-04-02:
I close issue as no update

closed 2018-04-02 by luc-github