luc-github/Repetier-Firmware-4-Davinci #293: First configuration and first trys

Tibercad, 2018-04-01:
Hello,
i have flashed my Da Vinci Duo 2.0A (with Jumper J37) and load the Failsafe parameters over the Menue of then Printer. Then i connect the Printer to the PC an set up the Printer Options to Davinci and the Parameter for connection. He will connect, so far so good. When i go on the right side of Repetier Host in the Field "Manual Control" he shows connctet. Here are the List of the liveparameters like Speed, Fan Speed and temperature. When i click on HomeX and HomeY the Nozzles move to right and away from me. When ich click HomeZ the Bed moves up. My first try of all without the Repetier Host in the printer directly is "Move Bed down" but the Bed moves up until the Nozzels hit the Bed. I had my hand on the on the Powerbutton to prevent a damage and switchted it imitiatly off. Must i switch the directions for all three directions? When i click in manual move nozzle Area in Repetier then the Nozzles move always in the other direction of the arrow of the Function in Repetier Host. Without them i dont see the option to calibrate the Bed correctly.
The Print Area in Repetier Host shows in my opionion false Parameter. Width(x)=200(!), deepth(y)=200(ok) and Height(z)=150(!). This must be x=150, y=200 and z=200 in my Opinion.

Sorry for my bad english.
Greetings
Tibercad

luc-github, 2018-04-01:
I am not sure what you want because I see only one question.

You must set repetier host by yourself - it is not a plug and play configuration - so yes setting are wrong because they are default settings and not for your DUO

about Fan - davinci fan cannot be controled by defaut - the fan are designed to cool extruder and start at 50C - that said if you want to setup a fan for the PLA because DUO has 2 fan - there is an option to repurpose one fan to be controled with M108/M107 and so by UI  - but you must also change fans position for the new functions : 1 fan to cooldown both extruder and one fan set on a fan duck for PLA (never use it on ABS) here an example : https://www.thingiverse.com/thing:1396452

Repetier is not XYZ software you have more flexibility but this also means you are in charge of all settings

Tibercad, 2018-04-01:
Hello Luc,
must i invert the directions of x,y and z?
I have thought the the fan2 starts like fan1 too. Ok, i will test it.
In the past i have used the auto calibration of the printer with the 3 heights and the adjustment with the 3 screws. It was easy. Is the adustment in Repetier like the adjustment in XYZ?
I think, the nose who recognised the contact hits not the contact area.

luc-github, 2018-04-01:
I do not use repetier host for years so you better read their documentation.
After flashing you must redo manual bed leveling as wrote in read me, only after you can do autoleveling. Do not compare XYZ and repetier, they have different approach.
Printer menu has manual bed leveling wizzard, it follow standard 3d printers procedure with paper sheet, when bed will be properly leveled, the autolevel contact should work

Tibercad, 2018-04-01:
Is the manual Bed levelling only in tue Printer or via Software from PC?

luc-github, 2018-04-01:
Sorry I do not know if repetier host has bed level wizard

luc-github, 2018-04-07:
Any update ? Can issue be closed ?

Tibercad, 2018-04-07:
I have levelled the printer with the automatic. Bevor i habe flashed the printer i have leveld it again with the menu on printer. After the flashing i habe testet the bed. A little bit move one screew and it was ok. All three heights are nearly the same (Z1: -6.079, Z2: -6.089, Z3: -6.84).
Now i have the Problem that the Nozzle is lower than the bed. For this Issue i have written in the Repetier Forum. [https://forum.repetier.com/discussion/5125/nozzle-crashs-into-bed#latest](url).
If you have an Idea to this Problem, pleas write me.
Greetings
Tibercad

luc-github, 2018-04-07:
as I wrote you must do manual bed leveling with screws then auto, your zmin should be 0 but it is -6,xxx which is not correct

I suggest you to watch videos about how to do bed leveling - this is not limited to Davinci, method is for all 3d printer.

I close issue as not related to FW

closed 2018-04-07 by luc-github