luc-github/Repetier-Firmware-4-Davinci #297: Variants.cpp does not exist in Program(x86) to be updated

Power3DPrinting, 2018-05-02:
Having trouble uploading the repetier firmware to an xyz davinci 1.0A with a version 2.0 board. No matter where I save the USB and Variants files in Arduino 1.8.0 and 1.6.9 when I upload to the 1.6.8 Due board in the instructions via a native USB port on my port 10 the board locks up and gives an error to my computer where it cannot connect via USB. The LCD is not any better with two rows of black rectangles and no response via the buttons.

Where in the world do I save these files in step 4 of the instructions? I found the USB.cpp under hardware, avr, and cores, which although it does not match up with the tree of files that it comes with off of github I just overwrote the existing one manually. The Variants I had more trouble with so I pasted in the Due file that it was in under the variants folder since there was no existing variants folder to be overwritten. Many thanks if any one can help me out with this!

Also, if anyone is familiar with how to wire a RAMPS 1.4 to an xyz davinci 1.0A, that would be a good back up plan to study. Not entirely sure how to wire things like the end stops since no signal wire is identified for the optical endstops, the heated bed has 4 wires, and I'm not entirely sure how to wire an autobed leveling system.

DocConnors, 2018-05-17:
With this newer version of Arduino, the USB and Variants files are store in C:\Users\<you>AppData\Local\Arduino15....
Caught me off guard too.

luc-github, 2018-06-02:
Everything is in the [readme](https://github.com/luc-github/Repetier-Firmware-4-Davinci#installation) and in the [FAQ](https://github.com/luc-github/Repetier-Firmware-4-Davinci/issues?utf8=%E2%9C%93&q=is%3Aclosed+label%3AFAQ+)  like [I cannot find variant.cpp/usbcore.cpp](https://github.com/luc-github/Repetier-Firmware-4-Davinci/issues/114)

closed 2018-06-02 by luc-github