luc-github/Repetier-Firmware-4-Davinci #298: 1.0a Fan1 won't spin at all.

DocConnors, 2018-05-16:
I've recently upgraded to a E3D V6, with it, i've wired in the 12V fan straight tot he power supply. No problem so far. I'm trying to fit a 40mm 5V fan to a print cooler I've fitted  and wanted to use the stock fan pins on the assembly board previously used by the original fan. No matter what combination of "REPURPOSE_FAN_TO_COOL_EXTRUSIONS", redefining pins or "FEATURE_FAN_CONTROL", the fan refuses to spin and the pin gives out 4.2V, whether on or off.

The fans been checked and is working fine with an external power supply, im at a loss.

luc-github, 2018-06-02:
if you disable REPURPOSE_FAN_TO_COOL_EXTRUSIONS like it was originaly - does you fan spin when you heat extruder as before ?

DocConnors, 2018-06-02:
It does not

luc-github, 2018-06-02:
So there is an issue with your wiring - as before it was working

DocConnors, 2018-06-02:
What I don't understand is it constantly reads 4.3-4.8V from the fan pins, irellevent of settings. But when the fan is plugged in, voltage drops to 1-2V.

luc-github, 2018-06-02:
I am not electronician so I can't give the exact root cause, I just can guess.

I only can say that  if you put FW in original state and fan is not spinning  like before - issue is definitly hardware:
 bad wiring if you did not wire like it was, or board is damaged,  or there is a short somewhere

The fact is Davinci is very fragile and easy  to get damaged  by short - so during your mod you may damage something

Sorry I cannot say more

DocConnors, 2018-06-17:
Nope, symptoms persist. I've opted for a manual switch

luc-github, 2018-07-22:
Ok so I close as hardware issue as original behavior is not working

closed 2018-07-22 by luc-github