luc-github/Repetier-Firmware-4-Davinci #299: Wrong temperature indication

Pascalmarin, 2018-05-29:
Hi,
I had install 0.91 version without issues along two years. Now I make a bowden conversion in my Davinci 1.0 with E3D V6 and I need to invert the extruder motor rotation. I download and configure the 0.92 version and install it. After a little print test, the printer reboot itself and the temperature indicator in the screen are 172/0 for extruder and 114/0 in the bed. The extruder temperature begins until 220ºC but the extruder and the bed are cold. If I try to print, the printer heat ups the extruder, but it does not begin to print. If I try to load or unload filament, the printer reboot.
I installed many times the Repetier firmware with same results. I Test with unplug the extruder and the thermistor sensor with the same result, have the same value in screen.
Any ideas about ?? I, m desperate?? :-D

Thanks

luc-github, 2018-05-29:
what thermistor do you use ? stock or E3D one ?
what change did you do in 0.92 FW ? 
Did you updated arduino for 0.92 ?

Pascalmarin, 2018-05-30:
Hi
**what thermistor do you use ? stock or E3D one ?**
The E3D V6 thermistor.

**what change did you do in 0.92 FW ?**
I activated the control fan:
#define REPURPOSE_FAN_TO_COOL_EXTRUSIONS 1
Inverted the extruder motor rotation:
// set to false/true for normal / inverse direction
#if DAVINCI==0
#define EXT0_INVERSE false
#endif
#if DAVINCI==1 || DAVINCI==4
#define EXT0_INVERSE true

**Did you updated arduino for 0.92 ?**
im used Arduino 1.8.5/Arduino DUE 1.6.8 in Mac
Im replaced USBCore.cpp and variant.cpp

Cheers.

P.

luc-github, 2018-05-30:
you know you cannot change thermistor without doing hardware rework ?
Check the FAQ : https://github.com/luc-github/Repetier-Firmware-4-Davinci/issues?utf8=%E2%9C%93&q=is%3Aclosed+label%3AFAQ+

and more specific topic : Using another thermistor than stock one https://github.com/luc-github/Repetier-Firmware-4-Davinci/issues/104

Pascalmarin, 2018-05-30:
Hello,
My Davinci worked perfectly with this Hotend with version 0.91, I mounted it following a tutorial and all works OK . I don't have enough electronic knowledge :-( but I try to change the resistors.
The strange thing is which the table value is incorrect too, and it haves the stock thermistor...
I,m lost!!!

Pascalmarin, 2018-05-30:
I read about this issue with startup temperatures about 19, 27, 25ºC, but my printer startup with 172ºC!!! in extruder and 114ºC in Bed
8-O

luc-github, 2018-05-30:
Yes table looks weird 
What branch of fw are you using?

luc-github, 2018-05-30:
what tutorial did you followed ? it may give a clue of the problem

Pascalmarin, 2018-05-30:
**what tutorial did you followed ? it may give a clue of the problem**
http://www.soliforum.com/topic/16731/howto-install-repetier092-on-da-vinci-10a-20a-updated-3/

**What branch of fw are you using?**
Im install 0.92.10
https://github.com/luc-github/Repetier-Firmware-4-Davinci

luc-github, 2018-05-30:
I do not see what could be wrong

My Suggestion to debug where error could come, as Arduino IDE setup is already done (including Due Module):

1 - Deleted FW directory you downloaded and download it again from github
2 - Compile FW without changing anything in configuration.h as by default FW is for Davinci 1.0 first generation
3 - Flash Firmware and check temperature

Pascalmarin, 2018-05-30:
Hi,
I deleted FW, compile without changes and flash... the same result.
I think have a Hardware problem, but I don't know the way to get :-(
[image]

Pascalmarin, 2018-05-30:
Sorry, I don't use stock thermistor!!!!

Pascalmarin, 2018-05-30:
Hi,
I found a problem, the thermistor wire is peeled and probably broke. Is probable this makes a short-circuit and fried some components in motherboard.
Any idea where to look?

Thanks

[image]

luc-github, 2018-05-31:
a short will do stop you from heating - check the fuse state 
there are 3 red led aligned on board - they show if short happen usually , if one fuse blow you need to replace

I saw oher people got this on forums (https://forum.voltivo.com/showthread.php?tid=15029&highlight=temperature and https://forum.voltivo.com/showthread.php?tid=10812&highlight=temperature) but cannot seehow they solved it
Voltivo forum is now dead so you can search on it but I suggest you to go to soliforum on davinci section and check even ask - they are still active and they may help you

Sorry cannot help more I am not electronician and knowing Davinci is very fragile I neved modded it and never got issue

Pascalmarin, 2018-05-31:
Thanks for your help!!
Now begins (or continue) the investigation work ;-D

Regards

P.D. I putted the Spanish translation in the the right Issue.

luc-github, 2018-05-31:
Good luck 
Yes saw it - will integrate soon - thank you

closed 2018-05-31 by luc-github