luc-github/Repetier-Firmware-4-Davinci #301: filament change issue

tecan, 2018-06-18:
when the printer runs out of filament during the print it would be nice if it reversed out the old filament before loading new filament. i had to recover a print because the z axis maxed out during the process too strange homing behavior.

tecan, 2018-06-20:
im not sure if once it hits the sensor its too late to put it back into the tube in the top ?

luc-github, 2018-07-22:
Yes indeed it is weird - loading/unloading is user responsability as many user case , no filament, clocked extruder/jam issue/ etc...
Stock sensor position is also little bit too close to extruder it should be more far to allow to take filament by hand.

tecan, 2018-07-24:
i wonder if we can use the top door sensor for an extra filament detector somehow.

luc-github, 2018-07-24:
you can use :
the Jam sensor pin which is not used
the door sensor pin 
the led pin - and connect led to power
the unpopulated connections of your board - some use pins

tecan, 2018-07-26:
https://www.thingiverse.com/thing:3021660 made one of these, works good

tecan, 2018-08-12:
https://cdn.thingiverse.com/renders/da/f5/24/e3/20/f2a1960ce7dafbbd2bfd6f6ee8af4442_preview_featured.JPG

luc-github, 2018-09-22:
I guess issue can be closed

closed 2018-09-22 by luc-github