luc-github/Repetier-Firmware-4-Davinci #303: windows 7 device not recognized davinci 1.0a

novyyyy, 2018-06-28:
Hi

been having a few problems with my davinci 1.0a. i formatted my laptop that runs my server for my printer reinstalled windows 7 x64 and now i cannot connect to my printer. i have reset my printer j37 and have two black bars on lcd display ready to flash .

i followed a few youtube videos tutorials and info from google using older versions of arduino etc and i keep running in to this same problem with device not recognized .

after laptop format windows  it can see 3dprinter as bossa and as soon as i compile and upload, bossa disappears and i get device not recognized

any help appreciated

novyyyy, 2018-06-28:
EDIT:

i managed to get it working again following this guys tutorial https://www.youtube.com/watch?v=rjuCvlnpB7M&list=WL

ardueno version 1.5.8
FW version 0.92.3  dated 2015

luc-github, 2018-07-22:
sorry for late reply and happy you handled it, windows can be difficult sometimes 
Happy printing

closed 2018-07-22 by luc-github