luc-github/Repetier-Firmware-4-Davinci #304: Power on without filament

LynxLynx123, 2018-07-01:
Hi Luc,
Thanks for your magnificent firmware, but I spotted an issue. When I've got no filament loaded and power on the printer, the x and y motors turn on for like 1s in home direction. When the axis are already homed they start to jumping on belts. 
Greetings

luc-github, 2018-07-22:
the motors are not driven at start at alll - so be sure there is no init.g file on SD or your host does not have init script 
also you can verify issue by disabling Filament sensor in printer

LynxLynx123, 2018-07-22:
Yes, you are right. This problem is at host. I'm using repieter server and it looks like when it sends a message "Extruder has no filament ! Push OK to go to change filament" it also sends command to move x and y motor. Of course after disabling of sensor, problem do not occur, host is not informed about lack of filament.

luc-github, 2018-07-22:
Interresting - I did not know host get the "no filament" event

Thanks for sharing - and sorry for late answer - seems I did not received a lot of issue on my mail box

I close issue then - happy printing

closed 2018-07-22 by luc-github