luc-github/Repetier-Firmware-4-Davinci #305: davinci 1.0a Z axis not homing and probably reversed

iacopo, 2018-07-05:
I've just flashed my 1.0a having set it to:

`#define DAVINCI 1
#define MODEL  1`

after flashing I rebooted and stored failsafe to EEPROM

Now almost everything seems to work fine, for instance lights turn on and off as expected (they didn't when I accidentally flashed it with MODEL 0)
The only big problem I have is the Z axis.

It seems like it's working in reverse and it does not home when entering the command.
Actually whenever I hit the Home Z command, the print bed just goes down and down, never up, so it never reaches the home sensor...
How do I fix that? Any help is greatly appreciated

iacopo, 2018-07-23:
I found out something on my printer PCB was toasted so the Z axis homing sensor is always detected as triggered. That caused the misbehaving I noticed.
My solution was to change pins on repetier, I used the top door optical sensor pin for the z-home instead and it works like a charm.

luc-github, 2018-07-23:
Davinci board is very fragile - happy you find a workaround - Issue could not be software :wink: 
Happy printing ^_^

closed 2018-07-23 by luc-github