luc-github/Repetier-Firmware-4-Davinci #306: extra extruder support

tecan, 2018-07-09:
was wondering if it were possible to add an i2c interface for adding extra digital and analog pins for other extruders and sensors.

luc-github, 2018-07-22:
Did you check the EEPROM cartridge pins ? they are free and connected

tecan, 2018-07-23:
[image]

i had 3 wires soldered on already which pins are these here ?

luc-github, 2018-07-23:
VCC GND and Pin 115 ?
Checkhttps://github.com/luc-github/Repetier-Firmware-4-Davinci/blob/master/pinout.xlsx

tecan, 2018-07-23:
also are the io pins rated for 5v ? or like arduino due and only like 3.5V ?

luc-github, 2018-07-23:
vcc should be 3.3v

tecan, 2018-07-23:
oh i see on the board com1 and com2 ports. awesome going to give this a whirl today hopefully.

tecan, 2018-07-23:
[image]

tecan, 2018-07-24:
https://github.com/tecan/PCF8574-SWire-Arduino.git   had to modify it to use new pins. works at 3.3 volts with 8 new i/o pins

tecan, 2018-07-24:
https://github.com/tecan/Adafruit_SSD1306-SWire-Arduino also this one for extra displays

tecan, 2018-07-24:
can confirm it working on pins 115 116

luc-github, 2018-07-24:
You connected esp01 with oled ?

tecan, 2018-07-24:
esp8285 because of the powersave feature then to the i2c portion maybe. for now it just sits there unconnected. have a flashair card that just arrived i might play with first. i just put some of the test code to the setup and loop for now in repetier. will be digging more into it later hopefully to add new olcd support.

tecan, 2018-07-25:
https://github.com/tecan/PortExpander_I2C-SWire-Arduino   there we go, turns out this is better i2c library for PCF8574.    the other lib was doing the pin reads backwards
 it likes steady high read state then pull to ground for action,otherwise the pins could not trigger eachother.
https://github.com/skywodd/pcf8574_arduino_library/issues/14

luc-github, 2018-07-25:
this is the olded I use on ESP3D - easy to program

luc-github, 2018-09-22:
I guess issue can be closed

closed 2018-09-22 by luc-github