luc-github/Repetier-Firmware-4-Davinci #309: Cannot compile

OAAbaali2896, 2018-07-17:
Hi,
I cannot compile the code and the error message states 
**Board arduino_due_x (platform sam, package arduino) is unknown
Error compiling for board Arduino Due (Native USB Port).**
I saved my file, from installing, in desktop & I didn't change anything in it. I only replaced the hardware file from repetier to 
C:\Users\osama_gc25s5w\AppData\Local\Arduino15\packages\arduino\hardware
I followed the instructions and I appreciate with the help.

OAAbaali2896, 2018-07-17:
I can finally compile the code with the USB port attached. The issue was using 1.8.4 rather than 1.8.0, I don't know why.

luc-github, 2018-07-22:
Sorry for late reply and happy printing

closed 2018-07-22 by luc-github