luc-github/Repetier-Firmware-4-Davinci #311: Adjusting firmware for custom splitter board

ChunkBuster, 2018-08-07:
I managed to burn out the board that splits the ribbon cable by being a total Gibbon, and ive created a replacement one. Everything works as expected... Everything except for the Extruder temperatures reported by the firmware. As the old board had resistors, I'm sure that there is some offset that you added to the firmware to compensate for this, but I can't find it. Where is this offset in the firmware?

luc-github, 2018-08-15:
Hi what printer have you ? 1.0/2.0 or 1.0a/2.0a/AiO, boards are differents

There is no offset but a custom themistor table because stock thermistor is unknown and looks different between board generations - I am guessing you are talking about the daughter board where Jam sensor(s) (not currently used), filament sensor(s), fan(s), heat extruder(s) and thermistor(s) are connected

luc-github, 2018-09-22:
closing as no update

closed 2018-09-22 by luc-github