luc-github/Repetier-Firmware-4-Davinci #312: Cant install firmware due to this error

SplitX1, 2018-09-05:
Arduino: 1.8.6 (Windows 8.1), Board: "Arduino Due (Native USB Port)"

In file included from sketch\Repetier.h:572:0,

                 from sketch\ui.cpp:20:

uiconfig.h:231:31: error: pasting "/* PINK.1, 88, D_RS*/" and "_PORT" does not give a valid preprocessing token

 #define UI_DISPLAY_RS_PIN  8  // PINK.1, 88, D_RS

                               ^

sketch\HAL.h:170:49: note: in definition of macro '_WRITE'

 #define  _WRITE(port, v)   do { if (v) {DIO ##  port ## _PORT -> PIO_SODR = DIO ## port ## _PIN; } else {DIO ##  port ## _PORT->PIO_CODR = DIO ## port ## _PIN; }; } while (0)

                                                 ^

sketch\ui.cpp:743:5: note: in expansion of macro 'WRITE'

     WRITE(UI_DISPLAY_RS_PIN, rs);

     ^

sketch\ui.cpp:743:11: note: in expansion of macro 'UI_DISPLAY_RS_PIN'

     WRITE(UI_DISPLAY_RS_PIN, rs);

           ^

uiconfig.h:231:31: error: pasting "/* PINK.1, 88, D_RS*/" and "_PIN" does not give a valid preprocessing token

 #define UI_DISPLAY_RS_PIN  8  // PINK.1, 88, D_RS

                               ^

sketch\HAL.h:170:84: note: in definition of macro '_WRITE'

 #define  _WRITE(port, v)   do { if (v) {DIO ##  port ## _PORT -> PIO_SODR = DIO ## port ## _PIN; } else {DIO ##  port ## _PORT->PIO_CODR = DIO ## port ## _PIN; }; } while (0)

                                                                                    ^

sketch\ui.cpp:743:5: note: in expansion of macro 'WRITE'

     WRITE(UI_DISPLAY_RS_PIN, rs);

     ^

sketch\ui.cpp:743:11: note: in expansion of macro 'UI_DISPLAY_RS_PIN'

     WRITE(UI_DISPLAY_RS_PIN, rs);

           ^

uiconfig.h:231:31: error: pasting "/* PINK.1, 88, D_RS*/" and "_PORT" does not give a valid preprocessing token

 #define UI_DISPLAY_RS_PIN  8  // PINK.1, 88, D_RS

                               ^

sketch\HAL.h:170:114: note: in definition of macro '_WRITE'

 #define  _WRITE(port, v)   do { if (v) {DIO ##  port ## _PORT -> PIO_SODR = DIO ## port ## _PIN; } else {DIO ##  port ## _PORT->PIO_CODR = DIO ## port ## _PIN; }; } while (0)

                                                                                                                  ^

sketch\ui.cpp:743:5: note: in expansion of macro 'WRITE'

     WRITE(UI_DISPLAY_RS_PIN, rs);

[... 2025 lines cut ...]

 #define UI_DISPLAY_ENABLE_PIN         125  // PINK.3, 86, D_E

                                            ^

sketch\HAL.h:170:147: note: in definition of macro '_WRITE'

 #define  _WRITE(port, v)   do { if (v) {DIO ##  port ## _PORT -> PIO_SODR = DIO ## port ## _PIN; } else {DIO ##  port ## _PORT->PIO_CODR = DIO ## port ## _PIN; }; } while (0)

                                                                                                                                                   ^

sketch\ui.cpp:810:5: note: in expansion of macro 'WRITE'

     WRITE(UI_DISPLAY_ENABLE_PIN, LOW);

     ^

sketch\ui.cpp:810:11: note: in expansion of macro 'UI_DISPLAY_ENABLE_PIN'

     WRITE(UI_DISPLAY_ENABLE_PIN, LOW);

           ^

exit status 1
pasting "/* PINK.1, 88, D_RS*/" and "_PORT" does not give a valid preprocessing token

SplitX1, 2018-09-05:
Any kind of help would help thank you

axlslak, 2018-09-19:
Do not install new versions of Arduino. Use the recommended version in the Read.me and make sure you patch it and it will work.

luc-github, 2018-09-22:
yes -thanks @axlslak  that the correct answer 1.8.6 has several issues

Sorry for late answer seems I do not get any notification for this repo

closed 2018-09-22 by luc-github