luc-github/Repetier-Firmware-4-Davinci #315: No reaction on Da Vinci 2.0 with Repetier Firmware

thefelix93123, 2018-09-25:
Hi
I flashed my DaVinci 2.0 Due with the Repetier Firmware 0.92.10M. At first everything worked and on the display is written "printer ready". But if i want to move the axis or heat the extruder, there is nothing working. I can also connect with Repetier Host (V2.1.2) and there is also nothing working. But there are this errors:
08:38:26.237 : Printed filament:NANm Printing time:0 days 0 hours 29 min
08:38:26.238 : PrinterMode:FFF
08:38:26.258 : DebugLevel:6
08:38:26.259 : SelectExtruder:0
08:38:26.268 : DebugLevel:6
08:38:26.269 : SelectExtruder:0

does anybody know what the problem is? (and sorry for my bad english)

luc-github, 2018-09-25:
you did not applied the M502/M500 commands or load and save fail safe in printer menu this is a must

thefelix93123, 2018-09-25:
Ok thank you, now something is working.
but the Extruder left turns by loading in the wrong side, X and Y axis go wrong if they should go to home position and also very slowly and the Z axis don't do anything.  Do you know this problem?

luc-github, 2018-09-25:
Nope
How do you use printer? From printer ? or from repetier?

thefelix93123, 2018-09-25:
it doesen't matter. Both is same. If i move the axis manualy, X and Y are doing right, just by homing they are wrong. on printer and repetier. Z axis don't also work from printer and repetier

thefelix93123, 2018-09-25:
It's a 2.0a micro sd

luc-github, 2018-09-25:
Did you put model 1?

thefelix93123, 2018-09-25:
I'm sorry, I found the problem by the x and y axis. The endstop is broken / maybe the wire. I took the switch from the top door and then it worked.
But i don't know what the problem is by the z axis and also the extruder. But i think i repair first the endstops and then i try again.
Thank you very much for your help :)

luc-github, 2018-09-25:
Ok then I close issue - Open a new one if necessary
good luck

closed 2018-09-25 by luc-github