luc-github/Repetier-Firmware-4-Davinci #316: Z-Axis don't work

thefelix93123, 2018-09-27:
Hi
I repaired my printer, maybe you remember (#315 )
Now the X and Y- axis are working correctly, but the Z axis don't move. I took the plug from another axis and tryed to move the z axis, but also no reaction, so the hardware should be ok.

And the second problem is, the extruder 1 turns in the wrong direction.

it's a da vinci 2 duo, motherboard with jp37.

do you have a idea what the problem is?

luc-github, 2018-09-27:
No - the code did not changed since a while and was working

Is it a working 2.0A stock FW or a second hand ? I am surprised the axis end stop were broken so easily

I am guessing the second extruder is not wired properly, but this can be changed in FW and
I supect also  Z motor has also wiring issue

just in case export EEPROM settings using repetier host and post here

thefelix93123, 2018-09-27:
It was a second hand printer :/  
[image]
[image]
[image]
[image]
[image]
[image]

[https://drive.google.com/open?id=1ROW1zXEGR5bpaSmgq9Df2LaWdTPG5Ggz](url)   and this is the eeprom file

luc-github, 2018-09-27:
Yeah second and modifed printer 
Check the extruder wires / connector - I won't be surprised if the wires were cut and reassembled, in worst case just change direction in FW
for Z you need to investigate : 

Check Z motor wires you can check continuity between board connector and motor

I guess sensor are working as you can see the temperature but I would also suggest you to check if printer is heating bed / extruder 1 / extruder 2 you may have others surprises

luc-github, 2018-09-27:
if wires are Ok I would guess step drivers issue or sensor issue - you see check sensor state when you press key down on printer

thefelix93123, 2018-09-27:
the sensor is ok :/ i also tried another sensor..  on this board i can't change the driver. is there anywhere a fuse?  or need i a new board?

luc-github, 2018-09-27:
I do not think it is fuse issue 

I suggest to ask on forum some advice , like in http://www.soliforum.com/ section Davinci - this may happen to someone else - 
as it is hardware issue - here is not the best place to get help - I only have a duo first generation

thefelix93123, 2018-09-27:
ok, i will try it there
thank you very much

luc-github, 2018-09-27:
good luck

closed 2018-10-10 by luc-github