luc-github/Repetier-Firmware-4-Davinci #320: During calibration the text refers to "a sheet of paper" what is assumed

The-Experimentalist, 2018-10-06:
Hi Luc

During calibration the text refers to "a sheet of paper" what is assumed to be the thickness of the paper?

0.1mm?
0.2mm?

I use feeler gauges to calibrate the printer and I am customising my code again and would rather refer to a specific size than to a "a sheet of paper".

Obviously the "a sheet of paper" is assumed to be of a certain thickness and the calibration is based on that. I hope you can help.

All the best 
Ex.

limited660, 2018-10-07:
A sheet of paper is typically .004" (around .1mm) but as long as what you use is consistent it doesn't really matter as you can adjust Z offset to get the correct first layer.

The-Experimentalist, 2018-10-08:
Thanks, where do you configure Z offset?

limited660, 2018-10-08:
I adjust the Z offset in the Slicer.

The-Experimentalist, 2018-10-10:
Thanks, what Slicer do you use out of interest? I tend to use Cura that is built in to Repetier for convenience.

limited660, 2018-10-10:
I use Simplify3D, Kisslicer, and Slic3rPE

luc-github, 2018-10-13:
I guess issue can be closed

closed 2018-10-13 by luc-github