luc-github/Repetier-Firmware-4-Davinci #323: Support for Da Vinci Mini W

2xrgmdak, 2018-10-07:
is it possible for you to create firmware for the Da Vinci Mini W? I will be happy to test it, I'm just tired of paying to much for NFC protected filament...

luc-github, 2018-10-07:
Sorry I do not have such printer so will be hard for me to hack it.
but can you take picture of printer 's  board and specialy  get the mcu I can see see if it is possible at least
is W for wireless ?

2xrgmdak, 2018-10-08:
[image]
[image]

here is an image of the mother board, sorry its not my image... i stole it off google. I havnt had the chance to take my printer apart. Yes the W stands for wirless, i have never acctually set up the wifi on it, but there is the option. I am unsure what you mean by MCU

luc-github, 2018-10-08:
The big chip with Atmel on it . need more close picture of it

2xrgmdak, 2018-10-08:
Okay, I will look around on google.  If I can’t find one I will have to open my printer up and take a picture for you tomorrow.

2xrgmdak, 2018-10-08:
[image]
here you go

2xrgmdak, 2018-10-08:
[image]
here you go

luc-github, 2018-10-08:
Looks like same as jr and pro which need a lot of work to avoid hack they split temperature with another cpu than the atmel one which make hacking more difficult

luc-github, 2018-10-08:
The positive is seems no screen
I do not SD card reader so I guess use software host is mandatory
I do no see how the wifi works if no screen / sd

2xrgmdak, 2018-10-08:
You have to open the xyz slicer and connect it to WiFi, then you can just send .3W files to it.

2xrgmdak, 2018-10-08:
And yes, the xyz ware slicer is mandatory. But I have found I can send .gcode to it using “windows print 3D”

luc-github, 2018-10-08:
you mean you send 3W file then can disconnect ? or xyz slicer  stream the file ?
is there any SD card ?

2xrgmdak, 2018-10-08:
There is no sd card. Yes, once you send the gcode file to the printer you can disconnect it from the computer

luc-github, 2018-10-09:
It means there is some flash to store the file which is new feature

luc-github, 2018-12-09:
Close a user was deleted - so no more request

closed 2018-12-09 by luc-github