luc-github/Repetier-Firmware-4-Davinci #325: Anyone had a go at installing the E3D PT100 upgrade?

The-Experimentalist, 2018-10-16:
I was wondering if anyone had had a go at installing/using the E3D PT100 upgrade?:

[https://e3d-online.com/v6-pt100-sensor-kit](url)

If anyone has any guidance I would be grateful.

luc-github, 2018-10-26:
the analog pis are defined in variant.cpp check for `PIN_ATTR_ANALOG`
some are used by  SD card read reader
others may not be yet located on board

The-Experimentalist, 2018-11-08:
Sorry,  I have been out of the country on business and I have not had chance to respond. I will be trying to work on this very soon. I currently run an Da Vinci 1.0 AiO with an E3D V6 hotend which I am looking to upgrade to use the PT100 but I am also just about to have a Da Vinci 2.0A Duo delivered which I intend to heavily modify using the E3D product range. I intend to update this with a Cyclops and 2 PT100s if I can find enough spare IO to accommodate it.

The-Experimentalist, 2018-11-15:
I have started to map out all the pins and will keep you informed of progress. I am away again for a few days so it will take a while. As I see it I am trying to verify the functions of AD4, AD5, AD6, AD7, AD8 & AD10 from the pinout spreadsheet as they do not seem to be assigned. How do I cross reference the analogue pins defined in variant.cpp with where they are used in code?

luc-github, 2018-11-15:
the pin used are in userpins.h - but use AD value not pin - check variant.cpp to see what pin it is

The-Experimentalist, 2018-11-15:
Luc I am not sure I understand. 

I can see that the relationship between the pin number and the name is created in userpins.h
Taking the extruder 2 temperature pin as an example (PB18 | AD9 | 89) I find in variant.cpp:

  { PIOB, PIO_PB13X1_AD9,   ID_PIOB, PIO_INPUT,    PIO_DEFAULT, PIN_ATTR_ANALOG,                   ADC0,  ADC9,  NOT_ON_PWM,  NOT_ON_TIMER },  // PIN 127 - PB13 with enabled ACD AD9

. . . in userpins.h I see:

#define HEATER_2_PIN        16//HEATER_0_PIN //switch pin value

However in pins.h I see:

#define HEATER_2_PIN   9

Is this the link between the PB13 and the AD9 in pins.h ???

luc-github, 2018-11-15:
This is heater
#define HEATER_2_PIN        16
This is sensor
#define TEMP_2_PIN      13 // ADC channel #, not a PIN #
This is sensor
#define TEMP_0_PIN               9 // ADC channel #, not a PIN #

Heater: Pin 16
Pin 16 in variant.cpp 
{ PIOA, PIO_PA13A_TXD1,    ID_PIOA, PIO_PERIPH_A, PIO_DEFAULT, PIN_ATTR_DIGITAL,                  NO_ADC, NO_ADC, NOT_ON_PWM,  NOT_ON_TIMER },

Sensor2: ADC13
ADC13 in variant.cpp is actually pin 67
{ PIOB, PIO_PB16X1_DAC1,   ID_PIOB, PIO_INPUT,    PIO_DEFAULT, PIN_ATTR_ANALOG,                   ADC13,  DA1,    NOT_ON_PWM,  NOT_ON_TIMER }, // DAC1

Sensor 0: ADC9
ADC9 is pin 63
{ PIOB, PIO_PB18X1_AD11,   ID_PIOB, PIO_INPUT,    PIO_DEFAULT, PIN_ATTR_ANALOG,                   ADC9,   ADC11,  NOT_ON_PWM,  NOT_ON_TIMER }, // AD9

Do not check pins.h, Davinci only use userpins.h

The-Experimentalist, 2018-11-28:
Sorry, I've been side tracked by a new puppy! I have been tracing the tracks from the ARM processor trying to find which analogue pins are connected. Unfortunately every track I have traced so far ends going to a via and so to another layer on the PCB. When I get a bit more time I will complete this work.I think I will have to take the board out and painstakingly trace them using a multimeter. I have opted to go for the water cooled E3D Chimera, with twin Titan extruders in a Bowden setup using Capricorn Bowden Tubing and I have also opted to use 2 x PT100 temperature sensors with the amplifier boards. I really need to find those analogues! I am going to design the Chimera carriage and print it on my Da Vinci AiO, with Repetier off course :0) I attach a picture of the tracing I have done so far. If you zoom in you can see I have traced several pins and tracks with coloured dots. I'll keep you posted but this is one of many projects at the moment and so the updates will be a bit hit and miss. Thanks for your help and guidance so far.

The-Experimentalist, 2018-11-28:
Trying to have another look in between other jobs. As I see it the analogues are on Arduino pins 54-65 which map to the ARM pins 78-91.

Is that right???

The-Experimentalist, 2018-11-30:
Hopefully I will be able to spend a little more time on this project over the weekend, thanks for the information and help as ever.

jick666, 2019-03-04:
Has anyone had success in installing this?  I recently purchased a PT100 and could use a bit of guidance.

luc-github, 2019-03-18:
Any update on this ?

closed 2019-05-12 by luc-github